PaulWieland/ratgdo #95: Pinout Question

cer2225, 2023-09-30:
Since the MyQ issue is does not seem to be going away anytime soon, , I have decided to use ratgdo's solution (Thank you Paul) but I have a question that I hope someone can answer. I am going to buy the WEMOS D1 ESP8266 and just solder pin headers on myself but I am a bit confused looking at the wiring diagram on the project page. Can anyone tell me exactly which terminal numbers to use for the Obst/GND/Ctrl connections. I will just use a USB brick for power so I do not need to connect the 3.3V.

Thanks all

PaulWieland, 2023-09-30:
The pins are labeled on the shield. See https://paulwieland.github.io/ratgdo/03_wiring.html

cer2225, 2023-09-30:
True, but I am not going to use the shield.  I am just going to solder pin headers on myself.

PaulWieland, 2023-09-30:
That's not going to work...

closed 2023-09-30 by PaulWieland

cer2225, 2023-09-30:
Okay, so you are saying that without the shield, it will not work?