PaulWieland/ratgdo #96: Can't access webui (could before)

monkey-of-the-hill, 2023-09-30:
Can't access Web UI of previously working/configured device. I've reflashed it and still no access. Can connect to logs over USB and it shows it's connected to WiFi, router shows it connected but no UI or ping.

PaulWieland, 2023-09-30:
See [Disable OTA & Web Server Config Access](https://paulwieland.github.io/ratgdo/02_configuration.html). To reset it you have to fully erase the flash on the D1 Mini when you install the firmware.

If you have been using ratgdo with your opener, make sure you get the last rolling code and then use [set_code_counter](https://paulwieland.github.io/ratgdo/01_features.html) to set that value after you erase and reconfigure.

monkey-of-the-hill, 2023-10-01:
Never mind. Network issues. I'm a dope. Works fine. False alarm.

closed 2023-10-01 by PaulWieland