PaulWieland/ratgdo #97: Unable to Connect to PC

kschmidt319, 2023-10-01:
Admittedly, I am new to this ESP thing, so I'm likely doing something wrong. But I've tried 3 different computers, 2 windows and 1 mac, and I cannot get it to recognize that the device is plugged in. I've also tried multiple wires. 

The blue light blinks a couple of times, and then goes off. Not sure if that's by design or not. No option to connect appear in the configuration page. I've installed all of the recommended drivers, with no change. Even with the blue light off, it still seems to be putting out a wifi signal because I can see that as available on my phone. It does not register in the device manager as a device being plugged in.

What step am I missing??

PaulWieland, 2023-10-01:
On the Mac press `command + space` and search for `System Information` (or navigate Apple Menu -> About This Mac -> More Info -> System Report).

Choose `USB` in the left column. With the D1 Mini plugged in you should see `USB Serial Converter` in the device tree at the top. You can use `Command + R` to refresh the USB tree if you plug/unplug the USB cable.

Here's what mine shows:
[image]

kschmidt319, 2023-10-01:
Thank you for the quick reply, Paul. Unfortunately, I do not see that option when I plug in the device:

￼

sco0ts, 2023-10-01:
Possibly try some different UsB cords? I have some that are just power and not data. They came with various electronics over the years.

shred86, 2023-10-19:
Just receive my ratgdo and having the same issue using a Windows 11 laptop. I know the cable is good since I've used it before to flash another ESPHome device. When I connect the USB cable to the ratgdo, a blue light turns on for about a half a second then turns off. In the device manager, I can see a "USB Serial Converter" with no driver - I've tried installing all three suggested drivers but none of them seem to work. Any pointers on getting this driver installed?

bedge117, 2023-10-19:
> Just receive my ratgdo and having the same issue using a Windows 11 laptop. I know the cable is good since I've used it before to flash another ESPHome device. When I connect the USB cable to the [...]

I installed via USB -> UART.  I had no issue Installing by doing this method.  I installed via ESPHOME so I could input my SSID and PWD.  Allowed for a quick adoption and configuration to HA.  ESP8266MOD  Tx->Rx Rx->TX VCC->VCC GND->GND and Use the install bypass pin ->GND.

sco0ts, 2023-10-19:
Any chance you can try another computer? I had issue with one pc. Other was
fine

kschmidt319, 2023-10-19:
Unfortunately, I was never able to get around the issue. I tried several different computers, numerous wires, all had the same results. I just assumed maybe I got a dud or something because nothing I tried worked.

sco0ts, 2023-10-19:
Did you try removing the esp from the rat expansion board then connecting
to pc? Maybe that will yield different result

shred86, 2023-10-19:
I ended up getting it to work. Not entirely sure what fixed it but I started manually installing drivers using the "pick from a list" option and selecting "have disk". The problem is there's multiple versions of the three drivers provided so there's probably 20+ choices you can make. I just started trying one by one, all of which would install but only some of which would allow me to connect to the device using the web ESPhome installer, but the firmware installation was failing. At some point I went back and tried to update the driver again and when I clicked "pick from a list", there was a new driver that seemed to be provided by Windows. I installed that and now everything is working. 🤷

razorfrog, 2023-10-30:
I'm having this issue now. Wondering if it's a USBC converter issue? My macbook only has USBC ports, So I'm using Micro USB > USB A > USB C converter. No luck so far.

PaulWieland, 2023-10-30:
Could be the adapter or the cable. I use an Apple branded USB C to A adapter and then a USB A to Micro data cable. Some cheap cables are power only.

Does it show up in your system information.app under the USB tree? Open the USB tree, use command + R to refresh. Look at the tree before plugging in, then plug in and refresh. You should see a USB Serial Converter device show up. 

[image]

razorfrog, 2023-10-30:
> Could be the adapter or the cable. I use an Apple branded USB C to A adapter and then a USB A to Micro data cable. Some cheap cables are power only.
>
> Does it show up in your system information.app [...]

I tried two off-brand USBC converters with no lock. Thankfully my older Mac Mini has USBA and it worked there. I'm up and running! 

Any chance of the next version using USBC?

cheatoman, 2023-11-08:
I am having the same issue :( I'm on Windows 10 with the v2.5 board. Tried installing the FTDI VCP drivers.

When I plug in the board, the blue light blips for a second then turns off. It showed up in my Device Manager as such
[image]

Then I go try an assign a driver to it. I select from list and first select the type as **Ports (COM & LPT)**, then select Manufacturer FTDI and select the driver.

Now with the web flasher, it shows up as PORT 3. I connect, install, erase. Then it gets stuck at **Preparing installation** for a bit before finally giving me the message:

Installation failed
⚠️
Failed to initialize. Try resetting your device or holding the BOOT button while clicking INSTALL.

Please help :(

sco0ts, 2023-11-08:
Can you try another usb port, cord, and computer.

My desktop never worked for esp boards. But my laptop does. No idea why

tlsJP, 2023-11-08:
Platform: Win11

Device: ratgdo v2.0 with ESP8266 d1 mini lite (clone)

I can get the device to detect and connect using CH341SER driver.  I've tried multiple devices for the driver (CH3330, CH340, CH341, etc), all of which do seem to connect, but none of which are able to flash properly. I get the 'Installing' modal with the spinning circle that says 'Preparing installation' underneath, but then it errors out.

Error message: Failed to initialize. Try resetting your device or holding the BOOT button while clicking INSTALL

cheatoman, 2023-11-09:
> Can you try another usb port, cord, and computer. My desktop never worked for esp boards. But my laptop does. No idea why
On my laptop, Chrome won't even connect to the COM port :( I've also played around with some different cables, but no dice

tubeleak, 2023-11-10:
> Platform: Win11
>
> Device: ratgdo v2.0 with ESP8266 d1 mini lite (clone)
>
> I can get the device to detect and connect using CH341SER driver. I've tried multiple devices for the driver (CH3330, CH340, [...]

Did you ever get this resolved? I am experiencing the same issue.

deep-barot, 2023-11-12:
I am in the same boat.

bedge117, 2023-11-12:
For those experiencing the failed to initialize error, Try to install to ESPHOME like this:

- Click the New Device button in esphome.
- Name your device
- During the Installation step, choose "Skip this step."
- Select your device type "This isn't really important to install the correct board just yet.  Just pick anything."
- On the "Configuration Created!" step, choose "Skip."
- This will create a placeholder for your new device.  It will appear offline.  Click "Edit."
- Modify your yaml file by copying Paul's examples from git hub. [(https://github.com/RATGDO/esphome-ratgdo/tree/main/static)](url)
- Choose the appropriate yaml file from under the static folder.  (note it is important to review the full name.  "Choose correct version. E.g. Version; either v2board or v25 board.  Choose the correct Board E.g. Esp8266_d1... etc..."
- Copy the raw data and paste it into your yaml file in your esphome instance. (Keep in mind you need to keep your esphome specific settings i.e. API, WiFi, OTA(Password), and any other settings you wish to keep.)

`substitutions:
  id_prefix: ratgdov2
  friendly_name: "RATGDOv2"
  uart_tx_pin: D4
  uart_rx_pin: D2
  input_obst_pin: D7
  status_door_pin: D0
  status_obstruction_pin: D8
  dry_contact_open_pin: D5
  dry_contact_close_pin: D6
  dry_contact_light_pin: D3

web_server:

esphome:
  name: ${id_prefix}
  friendly_name: ${friendly_name}
  name_add_mac_suffix: true
  project:
    name: ratgdo.esphome
    version: "2.0"

esp32:
  board: wemos_d1_mini32

dashboard_import:
  package_import_url: github://ratgdo/esphome-ratgdo/v2board_esp32_d1_mini.yaml@main

packages:
  remote_package:
    url: https://github.com/ratgdo/esphome-ratgdo
    files: [base.yaml]
    refresh: 1s

time:
  - platform: homeassistant
    id: homeassistant_time
    
api:
  id: #your ID here
  
ota: #your ID here
improv_serial:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  
  ap:
    ssid: #fallback name
    password: #your password
logger:`

- Click Save, and Click Install
- Now click the "Manual download" option.
- Your code will now compile.  When it is time to download, click the "Modern Format"
- Navigate over to [adafruit.github.io/Adafruit_WebSerial_ESPTool/](url)
- Connect your device to your PC.  Hold down the boot button on the esp board, and plug it in. Click "Connect" in the upper right."
- If successful, leave all the default settings I.e. 115200 Baud (If you see "Try hard reset." in the logs, that's ok.  Don't worry about it.  You just need to see Connected successfully and the "Connect" button should now say "Disconnect."
- Next, Choose Erase.
- Don't do anything with your board here. Once Erase is successful, click the upper most "Choose a file..." button.
- Leave the offset at 0.  Navigate to the .bin file you saved earlier when clicking Modern Format.
- Click "Program."
- At this point, it will upload to the board.
- Wait until confirmation, and navigate back to your esphome instance.
- Unplug your board and plug back in. (Don't hold the boot button.)
- Once the device is connected to wifi, you will see the device show up as "Adopt."  
- Adopt the device and wait.  You may be promoted to "install" again, but since your OTA is set, just install wirelessly. (There are some settings that are over written with the way the Project is compiled.  It will prompt you to install if you changed the default name.)

- Final Step, Enjoy being Free from MyQ!!!!!!

tlsJP, 2023-11-14:
> Did you ever get this resolved? I am experiencing the same issue.

I didn't get it fixed; I'm taking a break from this lol

joergbattermann, 2023-11-15:
> For those experiencing the failed to initialize error, Try to install to ESPHOME like this:
> ...
> * Connect your device to your PC.  Hold down the **boot button** on the esp board, and plug it in. Click "Connect" in the upper right."

Thanks for the description, going to try that one! However / I must be blind, but do you happen to know where the boot button is on the 2.5 board / https://github.com/ratgdo/esphome-ratgdo/blob/main/static/v25board_esp8266_d1_mini_lite.png ? I can only see the Reset button on mine and see also the image.. is it on the underside of the pcb / _is_ there one at all?

bedge117, 2023-11-16:
On that particular board, there isn't a button for boot. However, all you
have to do is connect pin D3 to ground.  It is GPIO0 pulled low.

I have had it where you cannot use the USB at all. If worse comes to worse,
you may have to get a USB to TTL Module serial converter.

I think they're on Amazon for less than 5 bucks.