PaulWieland/ratgdo #98: Collision detected message; unit installed

greggitter, 2023-10-01:
Hi, I looked through all the issues and saw one related to this message where someone had not yet installed the unit and was getting this message, but I have the unit installed.

Soldered the D1 mini to the shield and flashed the board using chrome.  Adopted via ESPhome in HA, it compiled and uploaded.  Connected red/white/black to the GDO.  System-wise, I can ping the board and get diagnostics but it seemingly fails to communicate with my GDO.  I'm wondering if my head unit is not compatible.  These two openers were purchased in 2014 (one has a battery backup and [this one does not](https://www.amazon.com/gp/product/B00APL6Q0W/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1)) and the only mention of 2.0+ security in the manual is this:  "This garage door opener is compatible with MyQ® and Security+ 2.0® accessories."

Assumption is "should" be compatible (?).  So I am wondering if there is anything else I can test to see where the problem is.  I've verified a dozen times the wires are connected correctly...even tried reversing white and black and that prevents the unit from booting.

Thanks!

PaulWieland, 2023-10-01:
Please post a picture of your wiring as well as the model number of your wall control panel.

greggitter, 2023-10-01:
Thanks for the quick reply.  Initially the wiring was basically using the wire clamps on the GDO itself but I convinced myself I had a connection issue and wanted to verify using pigtails.

[image]
[image]
[image]

PaulWieland, 2023-10-01:
The wires are connected to the wrong terminals on the shield. Please look at the terminal labels on the backside where the QR code is.

greggitter, 2023-10-01:
Well that was easy.  I didn't see the small writing and was referring to wiring diagram and because I could see the antenna on the D1, I assumed I was looking from top down.  Thanks so much!  :)

closed 2023-10-01 by greggitter