PaulWieland/ratgdo #99: Door always shows open. Close button closes and opens.

sco0ts, 2023-10-01:
Just hooked up the ratgdo. Showing door state as always open. Showing light as always on. 

Down action makes the door close and open. 

Stop action does nothing.

sco0ts, 2023-10-01:
[image]
[image]

PaulWieland, 2023-10-01:
What’s your order number?

sco0ts, 2023-10-01:
*11ee5229778b9fc593a3ac1f6bbba828*

PaulWieland, 2023-10-01:
Sorry that’s not it. It would be on your receipt sent by square. I can also check by email address.

sco0ts, 2023-10-01:
The issue is with the item from my most recent order. I cannot find that
order number. This is the order number from a previous order if you are
able to look up my other orders
2033347675

PaulWieland, 2023-10-02:
I found it. Will send you a replacement as soon as possible.

PaulWieland, 2023-10-02:
Email sent

closed 2023-10-02 by PaulWieland

barthjp, 2023-10-19:
I'm having a similar issue. It looks like I'm not seeing any status updates on the door or light. The only status updates I'm getting are the availability and obstruction. Can you comment on how this issue was resolved?
[image]

Order# 11ee5af182dd9a8893a3ac1f6bbba828