PaulWieland/ratgdo #101: Does RatGDO work with 889LM?

dubhunter, 2023-10-03:
I just backordered 3 of your awesome products due to MyQ locking down their APIs.

I then realized I had added MyQ support to an older GDO with the 889LM. I am wondering if your product will work with this?

Thanks in advance, and fantastic work!

PaulWieland, 2023-10-03:
Progress is being made and I should have a solution pretty soon. Please standby!

dubhunter, 2023-10-03:
Awesome! Thanks again!!

txwindsurfer, 2023-10-03:
I have ordered 3 ratgdo.  One is installed on a relative new Chamberlain the other 2 are much older Chamberlain Formula 1 (8500?) from about 2002.  The new one works like a champ in HA.  Awesome work!  The other two appear to have the same issue as described in #62.  The MyQ controller is a myq-g0301-d on those old models.  I am hoping this new solution will work with the older lifts.  Thanks!

PaulWieland, 2023-10-06:
@dubhunter You might be able to help me out if you are willing to do some alpha testing. Does your opener have a purple learn button?

If it does, please flash [version 2.23a](https://paulwieland.github.io/ratgdo/flash.html) and then capture the serial monitor output as you cycle the door from closed, opening, open, closed.

This requires you to have your computer plugged into the D1 Mini to get the serial logs.
Also be aware that the door will be triggered if you flash or reboot the D1 Mini with it plugged into the shield. I'm working on a fix for this.

Please see my comment in #54 for more info.

dubhunter, 2023-10-06:
@PaulWieland I would be happy to, but I don't have my hardware yet (I just [back]ordered 10/2/23).

I think I have a purple-ish button (image attached). If you are able to send me one of the units early, I am happy to help.

[image]

PaulWieland, 2023-10-06:
That’s perfect. I’ll get you a board as soon as I can.

dubhunter, 2023-10-06:
@PaulWieland emailed you.

cliff85, 2023-10-07:
I have the PD210D SN 41ac050-1m with purple learn button. I tested with 2.23a. Light and obstruction are working but open/close door doesnt always work. I have execute the open or close command 2-3 times for it to trigger.  These are the logs from the console. 
[esp-web-tools-logs (1).txt](https://github.com/PaulWieland/ratgdo/files/12838493/esp-web-tools-logs.1.txt)

PaulWieland, 2023-10-07:
@cliff85 thanks for the log. Can you test the following:

using your wall panel only to control the door: start with door closed and wait ~20 seconds. open the door and let it fully open. wait ~20 seconds then close the door and let it fully close, wait another ~20 seconds.

It should capture the data that the GDO and the Wall Control are broadcasting during those cycles. I believe that the signal data for openers which use limit switches are different from those that use rotary encoders (for some reason).

cliff85, 2023-10-07:
I didnt see any updates to log/console when using the wall panel to open and close the door. These were my steps

Plug in ratgdo to PC, it opened the garage door on power
Connect to ratgdo console
wait 20s
Wall control to close garage door and wait 20s
Wall control to open garage door and wait 20s
Wall control to close garage door and wait 20s

[image]
[esp-web-tools-logs (2).txt](https://github.com/PaulWieland/ratgdo/files/12838620/esp-web-tools-logs.2.txt)
[image]

download logs

PaulWieland, 2023-10-07:
Wow interesting. Now I wonder if the protocol is determined by the wall panel. If you don't mind continuing to test this, I'll build a serial logger firmware that doesn't do anything except output the raw serial data...

cliff85, 2023-10-07:
Yes no problem I can help test.

PaulWieland, 2023-10-07:
Ok on the install page you will find a Serial Dump utility. Install that and then open the Logs & Console. You should see it spit out raw data to the console.

[image]

[image]

Sometimes the esp web tools console window is finicky and I have to click the reset device button to get it to show whats going on, but the download logs always seems to work even if nothing is shown in the window. Just beware that the door will move if you click reset/install/etc. Pull the safety release on the door trolly if you want.

Once you have it capturing data, repeat the steps with opening/closing the door while waiting 20 secs between.

cliff85, 2023-10-07:
I tried to reset it a couple times because looks like I am just getting FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 

Attached logs of same opening/closing the door while waiting 20 secs between

[esp-web-tools-logs (3).txt](https://github.com/PaulWieland/ratgdo/files/12838919/esp-web-tools-logs.3.txt)

PaulWieland, 2023-10-07:
Ahh ok. I think the FF is the gdo searching for a wall panel which it does after the power cycle.  

It takes up to a minute on my test rig before it starts outputting any data after a reset of the esp which pulls the serial line low long enough for the gdo to go into search mode. 

It’s also possible that yours might never output anything else since you have a much simpler wall panel. 

Let everything sit powered on for 5 minutes then check to see if the serial log includes anything new.

cliff85, 2023-10-07:
Same results. Gave it 5 minutes and tried to open/close but still FF.

[esp-web-tools-logs (5).txt](https://github.com/PaulWieland/ratgdo/files/12838966/esp-web-tools-logs.5.txt)

PaulWieland, 2023-10-08:
@cliff85 So your test confirms that the GDO looks for a wall panel and must detect one before it will report its status back on the serial line. I just pushed 2.24a which adds wall panel emulation.

In this version when ratgdo boots up it listens for a wall panel. If after ~35 seconds it doesn't hear from one then it switches to emulation mode which _should_ cause your GDO to start reporting its status. Hopefully then we can get the values for each of the four main door states we need.

Now I have no idea what will happen with your existing wall panel after the GDO switches to serial mode, so you will need to test that too.

Example:
```
QUEUE MSG SENT [test/GlassDoor/status/availability] 
online
Looking for wall panel...
Looking for wall panel...
Looking for wall panel...
Looking for wall panel...
Looking for wall panel...
Looking for wall panel...
Looking for wall panel...
Looking for wall panel...
Looking for wall panel...
Looking for wall panel...
Looking for wall panel...
Looking for wall panel...
Looking for wall panel...
Looking for wall panel...
Looking for wall panel...
Looking for wall panel...
Looking for wall panel...
Looking for wall panel...
No wall panel detected. Switching to emulation mode.
3535 35FF 3333 3826  | door state: 26 100110
3A59 Light state off
QUEUE MSG SENT [test/GlassDoor/status/light] 
off
3A59 3A59 3905 3826  | door state: 26 100110
3A59 3826  | door state: 26 100110
3A59 3905 3826  | door state: 26 100110
3A59 3905 3826  | door state: 26 100110
3A59 3905 3826  | door state: 26 100110
3A59 3905 3826  | door state: 26 100110
3A59 3905 3826  | door state: 26 100110
3A59 3905 3826  | door state: 26 100110
3A59 3905 3826  | door state: 26 100110
3A59 3905 3826  | door state: 26 100110
```

kmalinich, 2023-10-08:
I'm definitely willing to test as soon as my hardware arrives, but that won't be until later this month, I think. I have an RJO20, which from reading the issues on here seems to also be in the "half-breed" camp. Mine also triggers an open/close when shorting the wall panel wires together.

cliff85, 2023-10-09:
@PaulWieland attached logs with 2.24a version.
[esp-web-tools-logs (6).txt](https://github.com/PaulWieland/ratgdo/files/12845287/esp-web-tools-logs.6.txt)

PaulWieland, 2023-10-09:
It’s nearly there. I assume you cycled the door using your wall panel? It pulls the line low which is why there are a couple of unknown door states in the log. I think I can filter those out.

cliff85, 2023-10-09:
Yes I cycled the door using the wall panel.

cliff85, 2023-10-12:
@PaulWieland I tried today from home assistant to execute open/close but the door doesnt respond. The light button works and all the statuses are correct (open/close/light/obstruction). Wall panel works properly. 

[esp-web-tools-logs (7).txt](https://github.com/PaulWieland/ratgdo/files/12875761/esp-web-tools-logs.7.txt)

PaulWieland, 2023-10-12:
@cliff85  thank you. I guess what I'm going to do is use a low pulse with sec plus 1 instead of the serial command. Multiple people are reporting that the door sometimes doesn't respond when you try and open it but I can't replicate it on my test rig. I think the low pulse should work fine though. I'm making a couple of other changes and will have a new alpha out hopefully by the end of the day.

apaperclip, 2023-10-12:
I'm not sure if I've seen this.  Perhaps I have and I assume I missclicked.   Does the wall panel just send the single serial command to open close or does it blast it a few times?

PaulWieland, 2023-10-12:
I just realized that I made a mistake in the last alpha and that my test opener doesn't seem to care. It should be sending `0x30, 0x31, 0x31`.

Version 2.24a is missing the first `0x30`. Will be fixed for the next alpha.

txwindsurfer, 2023-10-17:
@PaulWieland I just finished flashing v2.2.4a to a RatGDO attached to an orange button Chamberlain.  Same outcome noted by @cliff85 in that everything works except the open/close commands.  Could the failure of the open and close commands be a rolling code counter sync issue?  Happy to help debug.

xtego, 2023-10-18:
Chiming in here.  I'm running 2.24a.  I can confirm that everything is working except open/close commands.  My wall panels are still working fine to open and close the door, but I've lost the ability to control the lights from the wall panels.  Not a big deal to me, but wasn't sure if that is expected behavior.

txwindsurfer, 2023-10-19:
Just checked and I can no longer control the lights from the wall panel either.

PaulWieland, 2023-10-19:
I think that the door not responding to open/close commands and losing wall panel control over the light are related issues. 

I have just pushed 2.26a which I hope will fix almost everything except for un-intended door triggering when rebooting the D1 Mini.

When installing 2.26a (even for secplus 2.0 users) choose the full erase option.

txwindsurfer, 2023-10-19:
@PaulWieland Wow you're the genius!  The open, close, and stop commands work!  The sensors all work, and the light will turn on and off with an MQTT command but not from the wall panel.  This bug with the light also exhibits this behavior: if you use MQTT to turn on or off the light then the open and close commands no longer work until you manually press the wall control button.  You don't have to wait until the door fully opens or closes to regain MQTT control, just push the wall button then you can use MQTT stop and open/close again.  Hope this helps.  I am available to assist if I can help.

PaulWieland, 2023-10-19:
Which wall control panel do you have? Can you paste the serial log? Will be helpful to note which event is what in the log.

txwindsurfer, 2023-10-19:
@PaulWieland I hope this is the type of serial log you are looking for.  Generated from the 1.0 serial dump.  
First log file is just open and close from the wall control panel:
[close then open with wall switch.txt](https://github.com/PaulWieland/ratgdo/files/13048164/close.then.open.with.wall.switch.txt)
This log file closes and opens with MQTT commands
[close then open with MQTT.txt](https://github.com/PaulWieland/ratgdo/files/13048128/close.then.open.with.MQTT.txt)
This log file starts with turning off the light then trying to open with MQTT command a few time and also trying the MQTT stop but nothing happens:
[turned off light tried to open with MQTT and tried to stop.txt](https://github.com/PaulWieland/ratgdo/files/13048138/turned.off.light.tried.to.open.with.MQTT.and.tried.to.stop.txt)
And this last log shows turning off the light, trying to open several times then opening with wall control panel then MQTT command to close works.
[full log.txt](https://github.com/PaulWieland/ratgdo/files/13048155/full.log.txt)

Wall control panel:
[image]
el:

PaulWieland, 2023-10-19:
@txwindsurfer thanks for your help. I just ordered a 78LM to test with.

PaulWieland, 2023-10-19:
Has anyone with a 889LM panel tested? Can you confirm the same behavior that @txwindsurfer is having with the 78LM?

xtego, 2023-10-20:
> Has anyone with a 889LM panel tested? Can you confirm the same behavior that @txwindsurfer is having with the 78LM?

I have a chamberlain 935cb.  I'm quite certain it's security+ 1.0.  I'll try to get you some logs shortly.  2.26a is giving me some interesting behavior, I think similar to @txwindsurfer.   I'll try to replicate his testing.

xtego, 2023-10-20:
Here is my testing. 
First, closed then opened to the door from the 935cb wall panel.  Worked as expected
[close and open from wall panel.txt](https://github.com/PaulWieland/ratgdo/files/13048961/close.and.open.from.wall.panel.txt)

Then opened and closed from MQTT.  Worked as expected.
[open and close from mqtt.txt](https://github.com/PaulWieland/ratgdo/files/13048970/open.and.close.from.mqtt.txt)

Finally, I cycled the light and opened the door from mqtt.  Then stopped from and closed the door from mqtt.  In this case, I had to hit open twice for it to work. 
[turn on light and off light then open stop close.txt](https://github.com/PaulWieland/ratgdo/files/13048979/turn.on.light.and.off.light.then.open.stop.close.txt)

dubhunter, 2023-10-20:
Still waiting for a RatGDO to test my 889LM.

gullygossner, 2023-10-21:
Just received my board and wired it up tonight. 

889LM connected to a RJ020C with a yellow learn button. 

- door opens when the d1 is powered up 
- the obstruction sensors work
- motion sensor isn't triggered 
- the door status doesn't appear to update correctly as it is shown as open when it is closed, cycling the opener doesn't change the status

kmalinich, 2023-10-21:
> Just received my board and wired it up tonight. 
> 
> 889LM connected to a RJ020C with a yellow learn button. 
> 
> - door opens when the d1 is powered up 
> - the obstruction sensors work
> - motion sensor isn't triggered 
> - the door status doesn't appear to update correctly as it is shown as open when it is closed, cycling the opener doesn't change the status 
> 
> I'm running the esphome version of the firmware

What version of the ratgdo software are you running on the d1?

txwindsurfer, 2023-10-21:
Use this link https://paulwieland.github.io/ratgdo/flash.html  and flash v2.26a using usb connection

gullygossner, 2023-10-21:
> Use this link https://paulwieland.github.io/ratgdo/flash.html and flash v2.26a using usb connection

I’m not able to see the firmware version in the esphome firmware but I’m guessing the esphome version is running v2.1 so you’re probably correct in suggesting this.

gullygossner, 2023-10-21:
> Use this link https://paulwieland.github.io/ratgdo/flash.html and flash v2.26a using usb connection

2.26a fixed the issue, security 1.0 was needed even though the opener has a yellow learn button.

PaulWieland, 2023-10-21:
What model number is your opener?

> 2.26a fixed the issue, security 1.0 was needed even though the opener has a yellow learn button.

gullygossner, 2023-10-21:
> What model number is your opener?
> […](#)
> 2.26a fixed the issue, security 1.0 was needed even though the opener has a yellow learn button.

it's A liftmaster RJ020C.

PaulWieland, 2023-10-25:
Thank you. Adding to our wiki.

closed 2023-11-01 by PaulWieland