PaulWieland/ratgdo #102: MQTT vs ESPHome

Ch0rchie, 2023-10-03:
Hi, what are your thoughts on RatGDO using MQTT vs. ESPHome? I already have other devices in ESP and haven't used MQTT yet, but wanted to see if there are any issues using ESPhome.

Thank you!!!

PaulWieland, 2023-10-03:
Theres nothing wrong with ESPHome, use that if you prefer.

Ch0rchie, 2023-10-03:
Thanks so much! I set up my ESP8266 today, just waiting on the shield. Do you have an estimate when orders from October will ship? Thanks!!

PaulWieland, 2023-10-06:
As long as nothing goes wrong, I should have all backorders shipped by the end of the month.

closed 2023-10-06 by PaulWieland

Ch0rchie, 2023-10-21:
Hi, just checking in to see if you have any updates on backorders? And for my shield order, will you be sending the v2.0 or 2.5?

Thank you! I am excited to get this installed!