PaulWieland/ratgdo #103: 3D Printed Enclosure

samwiseg0, 2023-10-04:
Not really an issue but I wanted to post that I created an enclosure with multiple mounting options. 

Fell free to add it to the documentation site

https://www.printables.com/model/602600-ratdgo-v2-pcb-enclosure

Thanks!

shred86, 2023-10-08:
Thanks for making this! I don't have a 3D printer so I'm looking at [online services](https://craftcloud3d.com) that will print it for me. Just to confirm, are the dimensions in your .stl files in millimeters? Just want to make sure I order the correct size.

samwiseg0, 2023-10-08:
> Thanks for making this! I don't have a 3D printer so I'm looking at [online services](https://craftcloud3d.com) that will print it for me. Just to confirm, are the dimensions in your .stl files in millimeters? Just want to make sure I order the correct size.

Yep they are in mm. If you are in the US/Canada I can print it for you for a nominal price. We can discuss things further via email if you would like. I can be reached at 3d@watch-tower.net

cdhgee, 2023-10-09:
@samwiseg0 What is the difference between the M3 version and the heat insert version? (Sorry if this is a dumb question, not worked with 3D prints before).

What's the approximate weight of the printed enclosure? I discovered my local library will print stuff for $0.10/gram, and I'm trying to figure out if that's a cost-effective option.

samwiseg0, 2023-10-09:
> @samwiseg0 What is the difference between the M3 version and the heat insert version? (Sorry if this is a dumb question, not worked with 3D prints before).
>
> What's the approximate weight of the [...]

Means that you can screw the screws directly into plastic on the m3 direct screw version vs using [heat set inserts](https://youtu.be/P7nHyI1TwKY?si=xllExMYDAKMqO8W5&t=188). 

It is about 35g. Hopefully their printers are well tuned as it is not the easiest print considering the tight tolerances for the lid. 

If you want it fully assembled with all the hardware I can sell you one. If you have further questions you can use the email I provided above.

apaperclip, 2023-10-19:
I saw a comment at printables that the distance of the mounting holes has changed on newer ratgdos. My ratgdo has 16m spaced holes. Based on a quick inspect of the .stl on printable it seems this is also for 16mm spacing.  Is the etsy version the same?

samwiseg0, 2023-10-19:
> I saw a comment at printables that the distance of the mounting holes has changed on newer ratgdos. My ratgdo has 16m spaced holes. Based on a quick inspect of the .stl on printable it seems this is also for 16mm spacing. Is the etsy version the same?

I was just made aware of the issue yesterday. I have both versions available in my store. I just ask that you specify which one you have after you have placed the order.

samwiseg0, 2023-10-19:
@cdhgee Can you provide some clarity on the hole spacing distance? 
Was this an intentional change? Is this something that will continue to vary from batch to batch? I just want to make sure I can provide something that would be consistent from batch to batch.

samwiseg0, 2023-10-19:
@PaulWieland Can you provide some clarity on the hole spacing? 
Was this an intentional change? Is this something that will continue to vary from batch to batch? 

I just want to make sure I can provide something that would be consistent from batch to batch.

techtomydoor, 2023-10-19:
@samwiseg0 Thank you for designing this and making the stl's available! Printing now and can't wait to tidy up my ratgdo install.

NathanBlais, 2023-10-25:
Or you can always buy a mounting box https://www.amazon.com/gp/product/B0B2RHX8B3/?ie=UTF8&th=1

PaulWieland, 2023-10-26:
Sorry for the lack of response, have been rather... busy...

I'll get you guys some outline drawings as soon as I can. I'll open a separate repo up on github.com/ratgdo for 3d printed stuff.

samwiseg0, 2023-10-26:
> Sorry for the lack of response, have been rather... busy...
> 
> 
> 
> I'll get you guys some outline drawings as soon as I can. I'll open a separate repo up on github.com/ratgdo for 3d printed stuff.

Thank you. Much appreciated!

apaperclip, 2023-11-07:
@samwiseg0 

If you still want some measurements from a digital caliper on a 2.5 board let me know.

samwiseg0, 2023-11-07:
@apaperclip Yes. Can you email me `3d@watch-tower.net`

Thanks!