PaulWieland/ratgdo #106: Two garages - one works, one doesn't

cjcole8, 2023-10-07:
I suspect it has something to do with the rolling codes? I have no idea what to set it to if that's the problem.

I initially had ratgdo installed in Garage A, it worked fine. I moved it to Garage B (it was harder to get to so I wanted to get it out of the way). A few weeks later, I got a new ratgdo device, and installed it back to Garage A. It falsely says there's an obstruction, and none of the commands work.

The control panel on the wall is 882LMW.
The actual unit connected to the ceiling is LiftMaster 7675.

Here's a log:
```
QUEUE MSG ARRIVED [ratgdo2/command/light] 
on
MQTT: turn the light on
rolling code for 12|light : 550100442FFFDDFDF7FF7F4032D10533737DBA
Write successful
QUEUE MSG ARRIVED [ratgdo2/command/light] 
off
MQTT: turn the light off
rolling code for 13|light : 550100420964B04B2C9259211068129589E4B1
Write successful
QUEUE MSG ARRIVED [ratgdo2/command/light] 
on
MQTT: turn the light on
rolling code for 14|light : 55010099208203208048049014DA08B53B3DBC
Write successful
QUEUE MSG ARRIVED [ratgdo2/command/door] 
open
MQTT: open the door
rolling code for 15|door1 : 5501009820804A041201278106434895A92498
rolling code for 15|door2 : 5501009820804A041201258106434895A93498
Write successful
QUEUE MSG ARRIVED [ratgdo2/command/door] 
open
MQTT: open the door
rolling code for 16|door1 : 550100060B65BEDB6DB64E6034C849B5A96DBE
rolling code for 16|door2 : 550100060B65BEDB6DB64F6034C849B5A97DBE
Write successful
QUEUE MSG ARRIVED [ratgdo2/command/door] 
close
MQTT: close the door
rolling code for 17|door1 : 550100052FB6D36DB6DBEA51166BECD58D3483
rolling code for 17|door2 : 550100052FB6D36DB6DBEB51166BECD58D1483
Write successful
```

PaulWieland, 2023-10-07:
The false obstruction message has nothing to do with the door not responding to commands. Double check your wiring.

The door not responding is indeed explainable if you had a ratgdo board connected to that door, then replaced it with a new one. This is essentially the same as erasing the flash as the rolling code counter started over at 0. 

To fix it, manually set your rolling code counter to some number higher than the number of commands you sent to the door with the original install (200? 500? 1000? pick something reasonable).

Instructions for the `set_code_counter` and `sync` are on [this page](https://paulwieland.github.io/ratgdo/01_features.html).

cjcole8, 2023-10-07:
[esp-web-tools-logs (2).txt](https://github.com/PaulWieland/ratgdo/files/12838100/esp-web-tools-logs.2.txt)

Using `set_code_counter` and `sync` aren't making a difference. I set it all the way up to 1000, and I'm positive I didn't use it that much before.

I included a picture of my wiring as well. It's kinda hard to see - I'm using two lengths of 18/2 thermostat wire. It's exactly the same setup as on my other garage, which works fine.

[image]
[image]

PaulWieland, 2023-10-07:
I think you have the obstruction and control wires reversed.

cjcole8, 2023-10-09:
Thanks. You were right - I rewired this thing a million times, I must have switched it the last time.

The up/down/light functions work now, but the Obstruction sensor is still showing constantly. I've tried making sure it is firmly inserted a few times, but it doesn't seem to make a difference.

PaulWieland, 2023-10-09:
It sounds like obstruction input is burned out now. The control line is 12v while the obstruction line is only 6v. When they were reversed the voltage on the obstruction input was twice as high as it was designed to be.

cjcole8, 2023-10-09:
Okay, to test this, I tried switching the two ratgdo units I have (but left the same esp8266 in place). Garage B still works fine (now with the ratgdo unit that you thought might have the obstruction line burned out), and Garage A is still acting very weird...but differently. It worked to make the garage door open on first try, but now the garage door movement and light is not working. However the obstruction sensor is working correctly (so both garages obstruction sensors are working).

cjcole8, 2023-10-10:
Actually, ONLY the garage up and the obstruction sensor seems to work.
Down, stop, and light do not.

cjcole8, 2023-10-14:
I've tried a few more things. Uninstalled and reinstalled it a few times. Put some different/thinner wiring in so multiple wires in one slot doesn't interfere. I noticed after a while that my control panel on the wall starts blinking when I try to use the ratgdo device. I have to push on it a bunch of times for it to finally start working again. I think the ratgdo and control panel are somehow interfering with each other.

PaulWieland, 2023-10-14:
They don't interfere with each other. When you flash ratgdo 2.0 while it is plugged into the GDO, the wall panel momentarily loses power and it has to resync. Thats what the flashing yellow light means. It doesn't have any consequence other than having to press the wall panel button once to sync it.