PaulWieland/ratgdo #108: Work with Liftmaster ML800EV

dermm, 2023-10-09:
Hello,
I just received my Ratgdo and tried it directly.

However, my GDO does not seem to work.
The obstacle detection works but nothing else.
No status and no open / close possible.
The ESPHome version sends a notification about an error in communication when starting.
I have tested both the normal MQTT SW and the ESPHome.

My GDO has a yellow round learn button.

What could be wrong?
Wiring:
[image]

PaulWieland, 2023-10-09:
I see two extra wires connected to the white and gray terminals on the GDO. I assume you have those wired to `wht/gnd` & `blk/obst` terminals on the shield? Where's the red/ctrl wire?

dermm, 2023-10-10:
The shield is connected via the Red/white/Black wire (Pos. 1 Red,2 White,3 Gray). 
On 2 and 3 is the obstacle detection connected.

limwakra, 2023-10-17:
Got the same motor here in the Netherlands, use the #0(green) for the red wire and don't use the red port. Got it working today.

dermm, 2023-10-18:
> Got the same motor here in the Netherlands, use the #0(green) for the red wire and don't use the red port. Got it working today.

Thank you very much, that was the trick!

closed 2023-10-18 by dermm