PaulWieland/ratgdo #109: 8550W Support

evanbills, 2023-10-09:
First off:  thank you so much for creating this product and project! 

I have a LM 8550W that I've connected up (photo attached).  I've also flashed v2.24a onto my v2.0 board recently ordered and received from you.  Controls don't seem to work at all, but I do get status updates (heartbeat?) on the MQ.

I'm guessing this is a Sec 1.0 opener, or hybrid, so I am not sure this will work but I wanted to try. (I was tired of getting ads in the Liftmaster app)

Please let me know if I have missed a configuration or if there is anything I can supply to assist troubleshooting.

[image]
[image]

PaulWieland, 2023-10-09:
Two things are wrong:

- You're using beta firmware. Please install v2.1
- The board is wired to the GDO incorrectly. Please check the wiring diagram/silkscreen on the back of the board

evanbills, 2023-10-09:
That did the trick!  Apologies, Paul.  I was following the wiring diagram [here](https://user-images.githubusercontent.com/4663918/235453980-04a642fa-a181-4297-b4f3-06e1315e02fa.png) vs. what's actually printed on the board :facepalm: .  

I can report that this works great with the 8550W.  Thank you again!

PaulWieland, 2023-10-09:
Thats the correct wiring diagram. It shows the back of the board (as indicated by the QR code and silk screen labels). Glad you got it working.

closed 2023-10-09 by PaulWieland