PaulWieland/ratgdo #110: ESP32-H2 and Thread?

lukescott, 2023-10-09:
Just wondering if you have considered Thread support w/ the ESP32-H2. Or other ways of connecting the Ratgdo without needing a WiFi Router as a middleman. 

(Also Matter, but that could probably be supported on the ESP8266 since Matter works over WiFi).

thenewwazoo, 2023-10-23:
FYI, Matter 1.2 does not have a garage door opener device type (it's [supposed to be coming in the future](https://www.theverge.com/22832127/matter-smart-home-products-thread-wifi-explainer)).

NathanBlais, 2023-10-25:
Also very interested in this. Even if matter does not support it yet, it would be nice to have the feature possibly added in the future.

lukescott, 2023-11-09:
There is a discussion of ESPHome supporting Thread/Zigbee:

https://github.com/esphome/feature-requests/issues/1397

Perhaps that could be used?