PaulWieland/ratgdo #111: LiftMaster 3280 and 41AB050-2 Logic Board

lukescott, 2023-10-10:
Looking at compatibility for a LiftMaster 3280. Mine has the logic board 41AB050-2. There are a few other  revisions: 41AB050, 41A5688, 41AB050‑2M.

https://www.liftmaster.com/receiver-logic-board-ac-belt/p/041AB050-2M

According to the manual it has Security+. I'm guessing 1.0? I think it might be older than some mentioned on the compatibility list because it has two boards. Wondering if it is compatible and if so where my solder points would be. (I have Ratgdo backordered)

Here are some pictures:

[image]
[image]
![IMG_3760](https://github.co
[image]
m/PaulWieland/ratgdo/assets/486224/4e32b355-797a-49ed-9367-538bbd7c0f74)

PaulWieland, 2023-10-16:
Looks like a purple learn button, in which case it is Security + 1.0. 

1.0 support is in alpha right now but will be released very soon.

lukescott, 2023-11-09:
@PaulWieland Just got the 2.5 board. A few questions:

- With the solder free installation, what features are supported with Security 1.0? (Door state / position?)
- Is the soldered installation only for Ratgdo 1.0 boards, meaning 2.5 is only solder free? (Besides the power source)
- Does MQTT firmware support Security 1.0? When will ESPHome support Security 1.0?

lukescott, 2023-11-09:
@apaperclip What the diff between the MQTT version and ESPHome? Any benefit of using one over the other?

apaperclip, 2023-11-09:
I can't speak to functional specifics since I'm 1.0 and can't use the esphome version yet, but generally:

- esphome will integrate with HA like other esphome devices. Its a great firmware for microcontrollers that can be used for many things. This means you do not need an mqtt broker
- esphome can run on an esp32 whereas i dont think native can
- I get the vibe that esphome version may haves some other quality of life features from perusing that repo's isues/prs
- If you are familiar with esphome, you can edit the config to do other things such as run wireguard, operate as a ble proxy, have onboard logic/automation, connect to other sensors such as temp/humidity, etc.
- the esphome version has contributions by some ha developers and I would expect it to be very high quality and cared for similarly to the native version

I'd like to go to the esphome version if/when it supports 1.0 simply because I like how esphome integrates with HA but honestly its more of a preference than any particular features. (for 1.0) The mqtt native version works so well for me that I may not!   

Hopefully that gives you a sense of the differences and what may work best for you.  Feel free to ask any other questions!  Maybe someone that can run sec+ 2.0 can chime in on more specifics.