PaulWieland/ratgdo #112: will it work with simpler legacy units?

thomasshawn, 2023-10-10:
Question: Can Ratgdo work on older less secure openers, as a bridge until they are replaced with newer ones?
\
background:
I have multiple openers. My newest ones are wall mounted liftmaster 8500 using a myQ bridge. My older 2 are **circa 2005 Model 696 CD/B legacy overhead door.** they are easily bypassed by simply jumping 2 wires on the back of the unit.

I currently have a myQ adapter on each of those units that functions very well to open and close and sense door position.-they have tilt sensors that attach to the door for position detection.

With all the recent myQ issues I'd like to just move over to Ratgdo for all my doors but i don't want to invest in a different system for the older doors and then have to reinvest in another when/if they fail since they are already almost 20 years old.

PaulWieland, 2023-10-11:
In the very near future it will work with good old fashioned openers as well. Two switches will be required to tell ratgdo what the door status is, one for open and one for closed. With those two switches it will report the four door states of `closed, opening, open, closing`.

Some openers have built in terminals for the limit switches. If your GDO doesn't then you will need to either open it and connect to the internal limit switches or buy two reed switches (or tilt sensors) and put them on the door track to sense when the door is fully up and fully down.

saxopwn, 2023-10-12:
I bought a ratgdo after the recent Home Assistant API issues- didn't do much research, just saw MyQ and pulled the trigger. I actually have a Marantec Synergy 260 wired to a Linear universal receiver that's connected to MyQ. Is there any way I can get this working now or will this be covered in the very near future? Thanks in advance.

[image]

PaulWieland, 2023-10-12:
It will work with that as long as your obstruction sensors operate on ~6v

you will have to add two reed switches to get door position. I’m going to post instructions as soon as I have the dry contact controls ready for testing.

PaulWieland, 2023-10-19:
beta support has been added for dry contact control in 2.26a. limit switch open should be wired to the `trigger - open` terminal, and limit switch closed wired to `trigger - closed`. Before connecting the obstruction sensors, make sure they operate on no more than 6vdc.

smiller171, 2023-10-20:
For my older Genie opener I've measured my obstruction sensor at 12.25v when obstructed and 10.85v when unobstructed. The contact sensors operate at 5v. I'm incredibly novice when it comes to working with electronics like this. Is it feasible for me to step down the voltage for the obstruction sensor to be able to hook it up to ratgdo, or would I need to just go without that functionality?

PaulWieland, 2023-10-21:
You would need a simple divider circuit to drop the obst signal voltage.

A 10kohm x 10kohm voltage divider should work since you know you have to halve the voltage.

rkkhanna, 2023-10-25:
@smiller171, I have a similar situation with my Genie GDO unit. Did you make any progress with voltage divider solution recommended by @PaulWieland?

> For my older Genie opener I've measured my obstruction sensor at 12.25v when obstructed and 10.85v when unobstructed. The contact sensors operate at 5v. I'm incredibly novice when it comes to [...]

PaulWieland, 2023-10-25:
Just to add a note, ratgdo will work fine without the obstruction sensor, it just wont be able to report the obstruction state. It's just a nice to have feature, but not required.

angrycatmeowmeow, 2023-10-25:
> Just to add a note, ratgdo will work fine without the obstruction sensor, it just wont be able to report the obstruction state. It's just a nice to have feature, but not required.

Excellent. I just ordered one for my super old craftsman dry contact opener (future proofing). I currently have an FP1 in the garage to turn/keep the lights on in the garage, but they're slow to sense presence. Now I can use the obstruction sensors of my GDO to trigger the main garage lights then have the FP1's take over.

smiller171, 2023-10-30:
> @smiller171, I have a similar situation with my Genie GDO unit. Did you make any progress with voltage divider solution recommended by @PaulWieland?

I don't have my ratgdo in hand yet, but I'm probably not going to bother with setting up the obstruction sensor initially. I don't currently have any plans for how I might use that data in automations, so it doesn't seem worth the extra effort.

thomasshawn, 2023-11-11:
Was about to place an order for my current units but the FAQ mentions issues with myq chamberlain 8500 units and older security despite the yellow learning button.

Does this still apply with ver 2.5out now?

PaulWieland, 2023-11-11:
I just clarified the FAQ.

For an 8500 you need a v2.5 and to use the security + 1.0 control protocol.

thomasshawn, 2023-11-13:
Do we follow the  the 1.0 connection solder/no solder Reed switch required guide on the 8500? Or just the programming protocol.?

thomasshawn, 2023-11-14:
Re reading my previous question it's a bit confusing.
On the 8500 you said use 1.0 security protocol but do we also need to have Reed switches for door position like with 1.0 or will all the other functions work like with 2.0 protocol?

PaulWieland, 2023-11-15:
reed switches are only used with the Dry Contact control protocol.