PaulWieland/ratgdo #113: Shields

amoisis, 2023-10-11:
Hi

This project is amazing! i am from australia and tried to order a shield, i am getting a backorder message and am unable to join the backorder list. 

Should i just be a bit more patient and will you post to australia?

Ariel

PaulWieland, 2023-10-11:
Hi Ariel,
I don't have square setup to accept AU addresses. Will need to verify compatibility with the openers they sell in AU as they are specific models that I don't have first hand experience with. I can check shipping cost but I need an address to do so. Please email me your address: ratgdo@gmail.com

audiato, 2023-10-12:
Hi Paul,
Another hopeful buyer from Aus here - I've also sent you an email to the above address with a couple of attached photos.
Pretty keen to get my hands on one if it looks to be compatible.

PaulWieland, 2023-10-13:
It’s really a gamble. It looks promising but I have no way of knowing for sure until we try it or someone with one of those openers has a signal analyzer

audiato, 2023-10-13:
Happy to give it a shot to confirm. Will email you my address.

bdinnocenzo, 2023-10-16:
Paul, you have done an amazing job with this project.  I would like to purchase 3 shields but the order form shows them out of stock and I can not place an order for them.  I am in the US.  Thanks in advance for your reply.

PaulWieland, 2023-10-16:
@bdinnocenzo you can still order even though they are on backorder. 

https://checkout.square.site/buy/S2C4K5QZO3Z65VLW3AEMW6IM

bdinnocenzo, 2023-10-16:
> @bdinnocenzo you can still order even though they are on backorder.
> 
> https://checkout.square.site/buy/S2C4K5QZO3Z65VLW3AEMW6IM

Paul,

I'm looking for the shield without the D1 Clone...I have a drawer full of them...and that is what cannot be ordered.

-Bill

PaulWieland, 2023-10-21:
Aus shipping now available. Compatibility with Australian security + 2.0 openers has also been confirmed.

closed 2023-10-21 by PaulWieland