PaulWieland/ratgdo #116: Chamberlain 41A5483-1C - Security+ 390Mhz - Will Ratgdo support this?

ahahamyan, 2023-10-11:
My garage dooor openers are from 2005 and have a red learn button indicating that it uses Security+ 390Mhz technology, see [Support](https://support.chamberlaingroup.com/s/article/How-To-Figure-Out-Which-LiftMaster-Remote-Works-For-Your-Garage-Door#Red_Orange) - I do use 3 button opener [Amazon.com](https://www.amazon.com/gp/product/B004RB73D0?th=1) which indicates Security+ 2.0 - does this mean that my opener is compatible with Ratgdo? My actual garage door opener board model is 41A5483-1C - [041A5483-1C | Receiver Logic Board, 390MHz | Discontinued | Parts | Chamberlain](https://www.chamberlain.com/receiver-logic-board-390mhz/p/041A5483-1C)

If not, what options do I have to control these reliably without having to rely on MyQ integration for HomeAssistant?

Thanks very much.

PaulWieland, 2023-10-11:
Is it really red? Or is it faded purple? based on the year my guess is that it's purple. That being said, chamberlain only lists green orange purple and yellow. 

Assuming its purple or orange, then it should be security + 1.0 wireline. Right now the security + 1.0 wireline protocol is available in an alpha state and it seems to be working pretty well but I have only been testing with purple learn button GDOs.

PaulWieland, 2023-10-11:
Have a look at this chart https://p.widencdn.net/nkm2rb/CX3131

ahahamyan, 2023-10-11:
I think it is red button or certainly looks it (see image below), and according to Chamberlain here is the info on red or orange learn buttons - https://support.chamberlaingroup.com/s/article/How-To-Figure-Out-Which-LiftMaster-Remote-Works-For-Your-Garage-Door#Red_Orange 

I also took a picture of the back panel of opener unit and in wall manual opener. 

I did look at the compatibility chart, but could not tell 100%, given it was manfuactured 02/05 in seems to fall into Security+ 390Mhz. Also the openers came with single button openers 971LM which appears to be 390mhz - https://www.amazon.com/Liftmaster-971LM-390Mhz-Garage-Remote/dp/B0002AU2II?th=1 

[image]
[image]

Thanks for your help. I would love to have Ratgdo work with these.

PaulWieland, 2023-10-11:
I would bet that it's compatible with security + 1.0 wireline but I can't guaranty it.

xtego, 2023-10-17:
Just learned the hard way (after buying 3 of your boards) that my doors are using security + 1.0.  Which effectively means I have 3 bricks over here.  Completely my fault, of course, for not researching closer.   With the updates you're working on for security + 1.0, will I need different hardware to make it work?  Can I assist with testing at all?

apaperclip, 2023-10-17:
try the 2.24 image.  that's the in progress 1.0 version

xtego, 2023-10-17:
> try the 2.24a image. that's the in progress 1.0 version

Awesome.  2.24a is mostly working for me.  With that installed, I can cycle the light and the door status, light status, and obstruction status are all properly reporting.  The only thing that isn't working is the opening and closing the door.  When I send the command from Home assistant, the opener just clicks for a second, but nothing happens.  Any ideas?

cliff85, 2023-10-17:
@xtego Same issue as https://github.com/PaulWieland/ratgdo/issues/101. Paul said he should have a fix in the next release.

PaulWieland, 2023-10-24:
Updated FAQ: https://paulwieland.github.io/ratgdo/09_faq.html

closed 2023-10-24 by PaulWieland

ahahamyan, 2023-10-24:
Does the ratgdo 2.5 board work with both 2.0 and 1.0 security+ openers? If I have a Security+ 1.0 opener now and get a 2.5 board to work with it, and later upgrade opener to Security+ 2 model, will the same ratgdo 2.5 board continue to work with newer opener?