PaulWieland/ratgdo #117: Will this work with the Genie 2128?

matchmee, 2023-10-12:
I have it set up right now with a MyQ hub but then I just noticed that it should have its own cloud smart system in it also. Wanting to take it off of anyone's radar and control locally.

PaulWieland, 2023-10-13:
Can you post a picture of your genie’s connection terminals? What happens if you short the wall control panel wires together?

matchmee, 2023-11-15:
[image]

so I received the new 2.5 board today and proceeded to connect it. I was not able to get any of the settings in ESPHome to work with it. When I flashed the MQTT firmware I was not able to get the 2.0 or the 1.0 versions to work. But with the beta of the dry contact, I am able to get the Safety sensor to report obstructions. The issue with the dry contact is that I’m not able to get the garage door buttons in HA to work. I can only get one of the sides to work (either up or down) but that side acts as a simple button that goes up or down or stop— much like the actual button on the wall. 

Do you have any advice on what could be done to make this work more closely how it is intended to work? When I short the wires on the wall control the door goes up or down or pauses.

PaulWieland, 2023-11-15:
> When I flashed the MQTT firmware I was not able to get the 2.0 or the 1.0 versions to work. But with the beta of the dry contact, I am able to get the Safety sensor to report obstructions

You installed the wrong firmware. From the flash page:

[image]

You are also going to need to wire limit switches. See the [wiring instructions](https://paulwieland.github.io/ratgdo/03_wiring.html) for dry contact doors.

closed 2023-11-15 by PaulWieland

matchmee, 2023-11-15:
Hey Paul, apologies if I wasn't being clear. I have flashed the board appropriately and followed the wiring instructions. I flashed it first the MQTT way off the very page you show in your comment and selecting the only version available for the 2.5 board. I could only get it to work when I selected the 'dry contact (beta)' option and restarted-- then I wiped the board and tried it with the ESPHome firmware in hopes maybe that would work better out of the box-- to no avail. 

I understand I could involve a more extensive wiring setup involving the magnetic sensors on the side of the door, but I was really hoping to avoid that somehow if Ratgdo was able to work with Genie doors more 'out of the box' in the way that the others do. Especially cuz it seems like the wiring and setup is fairly similar. I know there's another guy on here who's been testing on a different Genie door and has seen some success. 

In regards to wire limit switches, I do not see anything mentioned in your documentation that would help me understand what you mean by that? I see a photo of a genie door that has limit switches on it, but I do not see anything like that on my garage door model.