PaulWieland/ratgdo #119: Any work done with the LA400U gate opener using the K1D8388-1CC control board?

ratsputin, 2023-10-13:
I have a gate opener that uses Liftmaster's K1D8388-1CC control board.  There are no obvious ways to connect a Security+ 2.0 device, not by physical investigation nor in the manual.  The only supported connections are "single button press" connections.

I do see an unlabeled 4-pin connection on the board but haven't had any luck tracking down its nature.  I have to assume it's likely JTAG.

Has any work been done along these lines?

DysfunktionalSD, 2023-10-20:
I'm curious on this as well since I have the same setup... Also wondering if this device will support multiple garage doors through a single device like the myQ-G0401-ES does.

PaulWieland, 2023-11-01:
Duplicate #90

closed 2023-11-01 by PaulWieland