PaulWieland/ratgdo #122: Liftmaster LM100evf

limwakra, 2023-10-16:
Ratgdo not working on Liftmaster LM100evf. 

Paul requested to do some voltage measurements on the ports:
[image]

[image]
Ratgdo is connected red on red (1) white on white(2) black on 3.

limwakra, 2023-10-16:
Obstacle detection voltage when detecting fluctuates from 3 to 5,6vdc and when nothing is detected and sensor is just on 5,3vdc

PaulWieland, 2023-10-16:
Just to make sure I understand your table correctly. You checked voltage between all pins to build that matrix?

I would expect pins 2 & 3 to be ground and to be common with each other but your table shows a voltage difference of ~1.2vdc?

PaulWieland, 2023-10-16:
Okay I missed that both white terminals were `#2`. 

~10v is approximately what I would expect for the serial line, but its possible its sec + 1.0 wireline which is a different baud rate.

Assuming you flashed firmware v2.1? What does the serial log show?

limwakra, 2023-10-16:
Just found that between the - from the other smal connector and RED(1) is 11VDC.

[image]

limwakra, 2023-10-16:
[esp-web-tools-logs (2).txt](https://github.com/PaulWieland/ratgdo/files/12920632/esp-web-tools-logs.2.txt)
the log file. funny thing is that when i click download log file the door opens.

PaulWieland, 2023-10-16:
Okay then! It looks like its sec+ 1.0. Flash the 2.24a (alpha) and change the protocol to Security + 1.0.

limwakra, 2023-10-16:
Logs sec+ 1.0 [esp-web-tools-logs (3).txt](https://github.com/PaulWieland/ratgdo/files/12920854/esp-web-tools-logs.3.txt)
Not working differently then the 2.1v.

PaulWieland, 2023-10-16:
A bit surprising. Maybe there is no serial interface at all and only dry contacts, or its also quite possible that the door isnt responding to the wall panel emulator that I wrote (which is modeled after a North American wall panel). 

Do you know if there are any wireline wall panels available for your door opener?

limwakra, 2023-10-16:
Yes should be the liftmaster 78EV. I don't own a wall panel.

PaulWieland, 2023-10-16:
I think we're at an impasse until someone who owns one of those wall panels and a logic analyzer can have a look at whats going on.

I'm on the look out for one, but its a tall order since they aren't sold in North America.

limwakra, 2023-10-16:
When in look in the manual of the 78EV it is connected not to the red but to the 0 green. Maybe there is the serial connection? 
[image]

PaulWieland, 2023-10-16:
Try it! Check both sec+1 and 2 protocols.

Just dont hook ratgdo to that line with 24v on it...

limwakra, 2023-10-16:
I got to go to work, I try it tomorrow. Big cliffhanger now.

Scott--R, 2023-10-16:
One thing to note - in many countries outside North America these types of openers look similar, but they work with dry contact on the red and white terminals. Also, there are some wall panels that add MyQ to older style openers, so the opener itself is dry contact but adding the MyQ wall opener causes the opener to "hiccup" and not always respond to dry contact inputs from a parallel device such as the RATGDO.

In short, based on what I see in your picture, your opener should respond to a simple short circuit signal between the red and white terminals.

PaulWieland, 2023-10-16:
Yes thats confirmed that the red terminal is a dry contact. On sec plus 1.0 north american openers, the red terminal is dry contact but it is also a serial line. 

However it looks like on these ROW openers the green terminal `#0` might be the serial line (based on that wiring diagram that @limwakra  found). We will find out soon 🤞

limwakra, 2023-10-17:
Ok good news, connected the ratgdo red wire to #0. And everything works when selecting securtity 2.0+ Security 1 was not working. Light works, door open close works. [https://github.com/PaulWieland/ratgdo/issues/108](url) This might also be the solution for isue 108.

limwakra, 2023-10-17:
Same manual from the 78EV the Liftmaster LM3800W ratgdo should be connected like this:
[image]

And all the other motors:
[image]

limwakra, 2023-10-17:
I guess I can now flash it to the ESPhome version? Or wouldn't that work, jet?

PaulWieland, 2023-10-17:
It should work just fine. Current version requires you to manually set the code counter (or you can cycle through the codes manually if you have only sent a couple)

the serial log outputs the last used code counter with each code it sends to the door, but you can also just guess a slightly higher number. It’s explained on the esphome repo how to input the starting counter.

closed 2023-11-01 by PaulWieland