PaulWieland/ratgdo #123: feature idea - detect when closing is interrupted

apaperclip, 2023-10-16:
(story unrelated to ratgdo but is the origin of the ideas)
Recently when leaving the house I triggered the door to shut and I drove away.  What I didn't see is the  small (<1" diameter) stick that the door hit forcing it to go back up.  It was too small to obstruct the obstruction sensor. This gave me the idea for 2 features

1. Detect when the door is closing and the obstruction sensors are tripped.  When this happens the door reverses and the light will flash...three (i think) times. This is clearly an error condition so I would expect there to be some communication from the GDO to the wall panel.
2. Detect the condition from the story. The obstruction beam is not blocked but the door detects enough resistance while closing to reverse. I do not know if this causes any unique communication between the GDO and the wall panel that ratgdo could detect, but I'm hopeful.

I will capture a console output (2.24a) of both scenarios and share. I know 2.24a is under active development but I wanted to mention this sooner than later for your consideration. If needed I can grab serial data from a logic analyzer. 

Oh and could timestamps and be added logs to make it easier to attribute log lines to events?  

Thanks!

PaulWieland, 2023-10-16:
When closing is interrupted, an obstruction message is sent. 

The event sequence and status of closing, obstructed, opening, open will be sent over mqtt for the scenario you described.

apaperclip, 2023-10-16:
Thanks! I was able to grab a set of logs

For triggering the obstruction sensor while closing - from a steady state of closed.  Above I was guessing the light blinked 3 times, it's more like 10 and there are no light state changes so it must be some internal behavior.

I definitely see the obstruction like you mentioned but also something unknown. Maybe this corresponds to the 80/81 value?

```
A5C 3904 3A5C 3852  | door state: 52 1010010
3A5C 3904 3A5C 3852  | door state: 52 1010010
3A5C 3904 3A5C 3031 3138 3A5C 3904 3A5C 3804  | door state: 4 100
Door state closing
QUEUE MSG SENT [ratgdo/garage/status/door]
closing
3A5C 3904 3A5C 3804  | door state: 4 100
3A5C 3904 3A5C 3804  | door state: 4 100
3A5C 3904 3A5C 3804  | door state: 4 100
3A5C 3904 3A5C Obstruction status obstructed
QUEUE MSG SENT [ratgdo/garage/status/obstruction]
obstructed
3880  | door state: 80 10000000
Door state unknown
QUEUE MSG SENT [ratgdo/garage/status/door]
unknown
Obstruction status clear
QUEUE MSG SENT [ratgdo/garage/status/obstruction]
clear
3A5C 3904 3A5C 3881  | door state: 81 10000001
3A5C 3904 3A5C 3881  | door state: 81 10000001
3A5C 3904 3A5C 3881  | door state: 81 10000001
3A5C 3904 3A5C 3881  | door state: 81 10000001
3A5C 3904 3A5C 3881  | door state: 81 10000001
3A5C 3904 3A5C 3852  | door state: 52 1010010
Door state open
QUEUE MSG SENT [ratgdo/garage/status/door]
open
3A5C 3904 3A5C 3852  | door state: 52 1010010
3A5C 3904 3A5C 3852  | door state: 52 1010010
3A5C 3904 3A5C 3852  | door state: 52 1010010
3A5C 3900 3A5C 3852  | door state: 52 1010010
3A5C 3900 3A5C 3852  | door state: 52 1010010
```

These logs are the door hitting resistance and reversing - again an unknown and maybe the 51 value?

```
3A5C 3904 3A5C 3852  | door state: 52 1010010
3A5C 3904 3A5C 3852  | door state: 52 1010010
3A5C 3904 3A5C 3852  | door state: 52 1010010
3A5C 3904 3A5C 3852  | door state: 52 1010010
3A5C 3904 3031 313A 3804  | door state: 4 100
Door state closing
QUEUE MSG SENT [ratgdo/garage/status/door]
closing
3A5C 3904 3A5C 3804  | door state: 4 100
3A5C 3904 3A5C 3804  | door state: 4 100
3A5C 3904 3A5C 3804  | door state: 4 100
3A5C 3900 3A5C 3804  | door state: 4 100
3A5C 3900 3A5C 3804  | door state: 4 100
3A5C 3900 3A5C 3804  | door state: 4 100
3A5C 3900 3A5C 3804  | door state: 4 100
3A5C 3900 3A5C 3804  | door state: 4 100
3A5C 3900 3A5C 3804  | door state: 4 100
3A5C 3900 3A5C 3804  | door state: 4 100
3A5C 3900 3A5C 3804  | door state: 4 100
3A5C 3900 3A5C 3804  | door state: 4 100
3A5C 3900 3A5C 3804  | door state: 4 100
3A5C 3900 3A5C 3804  | door state: 4 100
3A5C 3900 3A5C 3804  | door state: 4 100
3A5C 3900 3A5C 3851  | door state: 51 1010001
Door state unknown
QUEUE MSG SENT [ratgdo/garage/status/door]
unknown
3A5C 3900 3A5C 3851  | door state: 51 1010001
3A5C 3900 3A5C 3851  | door state: 51 1010001
3A5C 3900 3A5C 3851  | door state: 51 1010001
3A5C 3900 3A5C 3851  | door state: 51 1010001
3A5C 3900 3A5C 3851  | door state: 51 1010001
3A5C 3900 3A5C 3851  | door state: 51 1010001
3A5C 3900 3A5C 3851  | door state: 51 1010001
3A5C 3900 3A5C 3851  | door state: 51 1010001
3A5C 3900 3A5C 3851  | door state: 51 1010001
3A5C 3900 3A5C 3852  | door state: 52 1010010
Door state open
QUEUE MSG SENT [ratgdo/garage/status/door]
open
3A5C 3900 3A5C 3852  | door state: 52 1010010
3A5C 3900 3A5C 3852  | door state: 52 1010010
```

I just realized there is a third condition - attempt to close while obstruction is present.   I'll grab that soon.

also - 
The unknown states and the syncing state are throwing warning logs in HA because they are not in the list of cover states (open/opening/closed/closing/stopped).  The states we may be seeing with these  conditions are useful to me, I'd like to set up an automation to alert me if either happens.

PaulWieland, 2023-10-18:
Okay I didn't realize you were on the alpha firmware. 
In any of the above testing did the door just stop and sit there or did it always reverse and open?

apaperclip, 2023-10-18:
it always reversed and reopened

apaperclip, 2023-10-19:
> I just realized there is a third condition - attempt to close while obstruction is present. I'll grab that soon.

With the GDO open, I blocked the obstruction sensor, and pressed the wall panel. I did the test twice. In hindsight I should have used the HA control so you could see when the close command came in.  It was right when you see the state 80 and unknown status.  The light again flashed 10 times for each test.

```
3A5C 3900 3A5C 3852  | door state: 52 1010010
3A5C 3900 3A5C 3852  | door state: 52 1010010
3A5C 3900 3A5C 3852  | door state: 52 1010010
3A5C 3900 3A5C 3852  | door state: 52 1010010
3A5C 3900 3A5C 3852  | door state: 52 1010010
3A5C 3900 3A5C 3852  | door state: 52 1010010
3A5C 3900 3A5C 3852  | door state: 52 1010010
3A5C 3900 3A5C 3852  | door state: 52 1010010
3A5C 3900 3A5C 3852  | door state: 52 1010010
3A5C 3900 3A5C 3852  | door state: 52 1010010
3A5C 3900 3A5C 3852  | door state: 52 1010010
3A5C 3900 3A5C 3852  | door state: 52 1010010
3A5C 3900 3A5C 3852  | door state: 52 1010010
Obstruction status obstructed
QUEUE MSG SENT [ratgdo/garage/status/obstruction] 
obstructed
3A5C Obstruction status clear
QUEUE MSG SENT [ratgdo/garage/status/obstruction] 
clear
Obstruction status obstructed
QUEUE MSG SENT [ratgdo/garage/status/obstruction] 
obstructed
3904 3A5C 3852  | door state: 52 1010010
3A5C 3904 3A5C 3852  | door state: 52 1010010
3A5C 3904 3A5C 3852  | door state: 52 1010010
3A5C 3904 3A5C 3852  | door state: 52 1010010
3A5C 3904 3A5C 3852  | door state: 52 1010010
3A5C 3039 3131 Door state syncing
QUEUE MSG SENT [ratgdo/garage/status/door] 
syncing
3A5C 3880  | door state: 80 10000000
Door state unknown
QUEUE MSG SENT [ratgdo/garage/status/door] 
unknown
3A5C 3904 3A5C 3852  | door state: 52 1010010
Door state open
QUEUE MSG SENT [ratgdo/garage/status/door] 
open
3A5C 3904 3A5C 3852  | door state: 52 1010010
3A5C 3904 3A5C 3852  | door state: 52 1010010
3A1C 3901 3A1C 3852  | door state: 52 1010010
3A1C 3901 3A1C 3852  | door state: 52 1010010
3A1C 3901 3A1C 3852  | door state: 52 1010010
3A1C 3901 3A1C 3852  | door state: 52 1010010
3A1C 3901 3A1C 3852  | door state: 52 1010010
3A1C 3901 3A1C 3852  | door state: 52 1010010
3A1C 3901 3A1C 3852  | door state: 52 1010010
3A1C 3901 3A1C 3852  | door state: 52 1010010
3A1C 3901 3A1C 3852  | door state: 52 1010010
3A1C 3901 3A1C 3852  | door state: 52 1010010
303A 3131 Door state syncing
QUEUE MSG SENT [ratgdo/garage/status/door] 
syncing
3901 3A1C 3880  | door state: 80 10000000
Door state unknown
QUEUE MSG SENT [ratgdo/garage/status/door] 
unknown
3A1C 3901 3A1C 3852  | door state: 52 1010010
Door state open
QUEUE MSG SENT [ratgdo/garage/status/door] 
open
3A1C 3901 3A1C 3852  | door state: 52 1010010
3A1C 3901 3A1C 3852  | door state: 52 1010010
3A1C 3901 3A1C 3852  | door state: 52 1010010
3A1C 3901 3A1C 3852  | door state: 52 1010010
3A1C 3901 3A1C 3852  | door state: 52 1010010
3A1C 3901 3A1C 3852  | door state: 52 1010010
3A1C 3901 3A1C 3852  | door state: 52 1010010
3A1C 3901 3A1C 3852  | door state: 52 1010010
3A1C 3901 3A1C 3852  | door state: 52 1010010
3A1C 3901 3A1C 3852  | door state: 52 1010010
3A1C 3901 3A1C 3852  | door state: 52 1010010
3A1C 3901 3A1C 3852  | door state: 52 1010010
3A1C 3901 3A1C 3852  | door state: 52 1010010
```

This is very similar to the obstruction-while-already-closing. Perhaps the 81 status is opening after obstruction.

PaulWieland, 2023-10-19:
Try 2.26a. it will treat those door status' as "opening" now.

apaperclip, 2023-10-19:
Will do - I saw your note about erasing flash before installing 2.26a which if I recall requires knowing a recent used code.  Is that still the case? if so I do not see a code in the log output above.

PaulWieland, 2023-10-19:
Not needed. just erase and install.

apaperclip, 2023-10-22:
good news! with 2.26a I no longer see unknown status in the logs for the 3 conditions above. Thanks for handling them!

However, there is still some  value that shows unknown.  I dont know what is causing it, the most recent occurrence was  while the gdo had been closed for ~40mins and no movement in the garage. I only know it happens from the HA system logs.   I will go through the log file and see if the occurrences relate to other actions.

Would you be willing  to expose the exception conditions (obstruction while closing, resistance while closing) somehow? I know the cover entity only has the 5 normal states, so perhaps there could be an error topic where closures fail due to obstruction or resistance?

PaulWieland, 2023-11-01:
I don't intend to do that because those statuses are tied to a specific model protocol.

closed 2023-11-01 by PaulWieland

apaperclip, 2023-11-01:
Thanks for considering!  I was thinking though, does sec+ 2.0 implementation not have the same risks?  Or does it handle the conditions in a completely different way.