PaulWieland/ratgdo #124: Collision detected issue

Totemus, 2023-10-16:
ESP Newby here. After flashing the board, I visit the device that shows eternal "Collision detected, waiting to send packets". When added to Home Assistant Obstruction sensor = Clear. The board can't operate the opener. Another ratgdo board connected to another opener works flawlessly, so I'm sure I wired the boards correctly.

J-Pipe, 2023-10-17:
Also seeing this when I flashed using the ESPHome flasher here: https://ratgdo.github.io/esphome-ratgdo/. I have not yet wired anything to the garage door opener yet.

J-Pipe, 2023-10-17:
Looks like I'm seeing this because I haven't hooked it up yet.
https://github.com/ratgdo/esphome-ratgdo/issues/47#issuecomment-1722371519

J-Pipe, 2023-10-17:
Actually, hooked it up today and still seeing the issue and nothing is working. Hmmmm.

Totemus, 2023-10-18:
> Looks like I'm seeing this because I haven't hooked it up yet.
> 
> https://github.com/ratgdo/esphome-ratgdo/issues/47#issuecomment-1722371519

Confirmed. Operator mistake.

closed 2023-10-18 by Totemus