PaulWieland/ratgdo #128: Order Status & Question

dapowers87, 2023-10-18:
Hi,

I'm sorry, I really tried to find this answer as best as I could before resorting to creating a ticket. 

I ordered a ratgdo a while back and gave payment. I never got an order confirmation email so I have no idea what the status is of the order, or even the order number.

Also, I ordered the `ratgdo v2.0 shield only` product. Did I buy the right product? I unfortunately don't understand the difference between `ratgdo v2.0 shield only` and `ratgdo v2.0 shield with ESP8266 D1 Clone`? I couldn't find any FAQ regarding this.

Thanks in advance! 

--David

apaperclip, 2023-10-19:
I can't speak to the order status but as for the two different options...  The ratgdo product that Paul designed and is offering is a board that works with an esp8266 microprocessor.  If you do not have one already, you can order the combo package that includes the esp8266 D1 clone.  

If you ordered just the shield and do not have an ESP already you can source one from a variety of places, you just need the D1 mini form-factor.  Amazon, aliexpress, etc should all have them depending on where you are.

*note it may work with the ESP32 microprocessor but I know the 8266 works and it's what he offers. We can get into those details if you'd like.

dapowers87, 2023-10-19:
Thank you for the clarification, @apaperclip! I just ordered an ESP8266 now.

closed 2023-10-19 by PaulWieland

dapowers87, 2023-10-19:
sending to your github email ending in sickdimension.com now