PaulWieland/ratgdo #129: Compatibility with LiftMaster 8500W

madansu, 2023-10-19:
Hi,

I have a Liftmaster 8500W garage door opener. 
Is the ratgdo v2 compatible with this model ?

Thanks,

apaperclip, 2023-10-19:
It _should_, but whether it uses the released secure+ 2.0 code, or the in-development secure+ 1.0 code will depend on the specific model and maybe the wall-panel.

Based on this link, the second row on the second page.  Is that your wall-panel?
https://www.windsordoor.com/assets/uploads/modules/accessory-compatibility-chart-91817-82926.pdf

closed 2023-10-19 by PaulWieland

NathanBlais, 2023-10-25:
Can I get some clarification on the wiring for this door? Since there are already wires inserted into the slots of the 8500W are we supposed to remove the wires connected to the existing wall opener or twist the ratGDO and existing wires together before inserting into the 8500W's wire slots?