PaulWieland/ratgdo #132: Do I need a wired wall button for this to work?

shanelord01, 2023-10-19:
Is there any way to get this working without a wired wall button installed?

I'm ion Australia and looking to use it with a [Merlin Commander Elite MS105MYQ](https://www.gomerlin.com.au/product/commander-elite-ms105myq/) but I have a wireless/battery powered Four Button Wireless Wall Mount Remote Control  (Security+ 2.0) (E148M).

[Page 6 of the manual](https://cgi.widen.net/s/rrbdxvzgxs/114a5180h) for the opener seems to line up with the required wiring.

Thanks.

shanelord01, 2023-10-19:
Damn - looking at [this post](https://github.com/PaulWieland/ratgdo/issues/8), even though my unit is a Security 2.0 device, it still seems to use dry contacts in the unit, so I am assuming ratgdo won't work for me? Main thing is I wanted to get this into MQTT and Home Assistant.

PaulWieland, 2023-10-19:
A wired wall panel is not required.

PaulWieland, 2023-10-19:
Have a look at https://github.com/PaulWieland/ratgdo/issues/122

closed 2023-10-19 by PaulWieland

shanelord01, 2023-10-19:
> Have a look at #122

Thanks! Exciting stuff. 

I’ll document with photos when I get them if it helps someone else.

limwakra, 2023-10-22:
Oh, that's interesting, they listed all the connections on page 7 of the Australia manual.
[image]

 You can connect your Ratgdo to Red/ctrl 0 and white ground at 2.

PaulWieland, 2023-10-26:
I've seen a number of different manuals and photos of this style opener (the ones with 14 terminals). Some have the first terminal labeled as 0, and some have it labeled as 1 🙈

But yes, what @limwakra said. You need the e-serial terminal and wire that to the red/ctrl terminal of the ratgdo control board.

shanelord01, 2023-10-28:
> The Ratgdo won't like the 30 Vdc 50 mA will probably kill the board, don't do that.

Thanks. I assumed as-is it wouldn't work - just wondering if there is something out there that would convert to USB friendly power. Be really nice to have it all in one powered by the door opener.

PaulWieland, 2023-10-28:
Yes thats enough. You can also pull the cover off the door opener unit and look for a 3.3v power source. Some are easily accessible.

shanelord01, 2023-10-28:
> Yes thats enough. You can also pull the cover off the door opener unit and look for a 3.3v power source. Some are easily accessible.

Thanks. I want to avoid opening it up as it’s still under warranty and gets serviced regularly.

shanelord01, 2023-11-14:
I know this is closed, but I have easily and successfully installed the 2.5i boards into my Merlin Door Openers here in Australia.

As you said for connections:

Merlin 0 - eserial (red/Ctrl)
Merlin 2 - ground (white/GND)

Installed ESPHome via the web installer (as I'm using with HA) and setup and running in mere minutes.

I'm now removing my door openers from my MyQ bridge, then removing the bridge from my home.

Thank you SO much.
Shane.