PaulWieland/ratgdo #134: Homebridge or scrypted plugin support?

pcm1ke, 2023-10-19:
This is not an issue, but I don’t see a discussion section.

With the latest spat of issues affecting third party plugins <> myq api, I’m guessing a solution like this may be starting to get even more traction. Has anyone come across a homebridge or scrypted plugin that brings this into HomeKit?

Thank you!

PaulWieland, 2023-10-19:
You could write a custom homebridge integration based on MQTT, but I _highly_ recommend using NodeRED instead. There is a homekit node for NodeRED that I find much easier to use and manage than homebridge.

[Check it out here](https://nrchkb.github.io)

rbclark, 2023-10-20:
As someone who already has everything working in Homebridge, I was going to first attempt integration using the [homebridge-mqttthing](https://github.com/arachnetech/homebridge-mqttthing/blob/master/docs/Accessories.md#garage-door-opener) plugin. Just ordered a ratgdo yesterday so as of right now I have no way to verify whether that is actually a solution or not.

pcm1ke, 2023-10-20:
Hi Robert,

Please come back and post to this thread after you get the board and try
this out. This was going to be my next step as well to finally put myQ api
behind me.

Thank you,
Michael

PaulWieland, 2023-10-20:
You can simulate ratgdo with any mqtt client in order to test the homebridge plug-in. 

Just publish the various status messages that ratgdo uses. They are all in the docs and source code but I can also give you examples from my nodered install if you want. 

Also fwiw you can run nodered along side home bridge. It’s just like pairing another bridge device in the Home.app

ChadGnad, 2023-10-30:
with the mqtt integration, can't you port from home assistant to HomeKit Bridge?  I use that for all my non HomeKit items and it integrates really well.

https://www.home-assistant.io/integrations/homekit

donavanbecker, 2023-10-30:
scrypted supports mqtt

dkerr64, 2023-11-01:
I have just ordered two ratgdo's for my garage... the recent myQ issues prompted me to search and find this.  I post here because I will want to integrate with Homebridge and while mqtt is one way to do that, it is probably best to have a dedicated plugin... something that can support the additional features like controlling the light, motion sensing, etc.

I'm willing to help... once the ratgdo arrives (they are backordered).  I have experience with Homebridge having written the [YoLink plugin](https://github.com/dkerr64/homebridge-yolink) in which I supported binding two of their devices together to simulate a garage door controller.

PaulWieland, 2023-11-01:
Actually there isn't much reason that I can see to have a HomeBridge version if you just have a native HomeKit version available. The [HomeKit version](https://github.com/ratgdo/homekit-ratgdo) is in the works and needs testers.

rbclark, 2023-11-01:
> Actually there isn't much reason that I can see to have a HomeBridge version if you just have a native HomeKit version available. The [HomeKit version](https://github.com/ratgdo/homekit-ratgdo) is in the works and needs testers.

That's a very good point, is there a writeup on how to set this up somewhere? Does it run on the actual ratgdo or does it require a separate server component?

dkerr64, 2023-11-01:
> Actually there isn't much reason that I can see to have a HomeBridge version if you just have a native HomeKit version available. The [HomeKit version](https://github.com/ratgdo/homekit-ratgdo) is in the works and needs testers.

Yes I agree.  I was not aware that there is native HomeKit support coming.  That is of course best

PaulWieland, 2023-11-01:
@rbclark  its native firmware for ratgdo. Onboarding will be done through ESPWeb Tools (to join wifi). All docs and issues related to the homekit version should be opened under the homekit-ratgdo repo since they will be specific to that firmware.

dkerr64, 2023-11-01:
> @rbclark its native firmware for ratgdo. Onboarding will be done through ESPWeb Tools (to join wifi). All docs and issues related to the homekit version should be opened under the homekit-ratgdo repo since they will be specific to that firmware.

Looking forward to it, once my order clears the backorder queue.  FYI, that repository is not accepting issues yet, understandable given early stage of development.  Also, what is the dev environment for it, does it require Apple Xcode?

nskerl, 2023-11-01:
> Actually there isn't much reason that I can see to have a HomeBridge version if you just have a native HomeKit version available. The [HomeKit version](https://github.com/ratgdo/homekit-ratgdo) is in the works and needs testers.

I am interested in joining as a tester. Ive already placed an order for a Ratgdo so happy to contribute once that arrives.

benyzboy8682, 2023-11-07:
I just ordered two ratgdo devices tonight. I would be happy to test the HomeKit version as well once my back order arrives.

cefoster0, 2023-11-07:
will the HomeKit firmware support both mqtt and native HomeKit? thinking of ordering as well but would like to integrate with both.

PaulWieland, 2023-11-07:
The HomeKit version will not include mqtt (it doesn't need to). That version is for people who just want ratgdo to show in their iOS ecosystem.

That doesn't mean you can't use HomeKit now (before the HK version is ready).

With platforms like NodeRED or HomeAssistant you can setup the integration between your home controller computer and ratgdo with either MQTT or ESPHome. Then on the platform you can expose that device however you like. So for example maybe you have an android phone and an iPhone in the same household - you can expose any "smart home" protocol that NodeRED or Home Assistant offers.

My personal installation uses MQTT <-> NodeRED <-> [NRCHKB](https://nrchkb.github.io/)

So NodeRED appears as a HomeKit hub in my iOS eco system with as many devices as I define in NodeRED with the NRCHKB node. NodeRED then updates those HomeKit nodes based on data it receives over MQTT.

It also appears as a Home Assistant device due to the auto discovery and they all work together in parallel and stay completely in sync because of the persistence of the MQTT data.

You can do the same basic thing with ESP Home & Home Assistant without NodeRED. The home automation platform you use is 100% personal preference. The thing that the homekit-ratgdo will do is enable people to interface their iOS devices with ratgdo without any need for a home controller. 

I hope that helps.