PaulWieland/ratgdo #135: Shipping to Canada

andornaut, 2023-10-19:
[The website](https://paulwieland.github.io/ratgdo/) mentions that shipping to Canada is available:
> ratgdo shields available to order Shields are available and shipping domestic USA via USPS. Shipping to Canada is also available, but costs $14.

When [checking out on Square](https://checkout.square.site/buy/X3QUEQGEQG4I6B24NC7EBHL3), however, Canada doesn't appear to be a supported shipping destination:
> This Address is not supported, please select another

[image]

It'd be great if shipping to Canada was offered!

Imitationman, 2023-10-20:
I don't think its the Canadian address that is the problem. I tried to place an order to Texas this evening and I get the same error that my address is not supported by Square. I think this is a Square issue.

PaulWieland, 2023-10-20:
Yes it is. I'm trying to figure out what is wrong. Their store management software is something else...

PaulWieland, 2023-10-20:
I think I have it fixed, please try again!

andornaut, 2023-10-20:
Worked. Thanks!

closed 2023-10-20 by andornaut