PaulWieland/ratgdo #141: Does ratgdo work with older units? Genie H6000A

aram535, 2023-10-20:
The H6000A has a manual light and door switch (wired), two wireless remotes, as well as inputs from obstruction sensors at the door.

The [pdf manual page 24](https://7941f249-da52-4516-9391-2e6ce33e2f43.usrfiles.com/ugd/3ef5a4_4ba22811d821444fa2bb0c5588c989ee.pdf) is the diagram of the connections.

smiller171, 2023-10-20:
Some of the docs haven't been updated yet (Specifically I noticed the Troubleshooting and FAQ page wasn't updated) but as of _yesterday_ older units are supported. 🎉

PaulWieland, 2023-10-21:
I need to find time to get the docs finished for dry contact openers. But the cliffs notes are:

door can be triggered

Door status can be determined using two limit switches which are wired to ratgdo’s trigger open/close terminals. Some gdos provide external terminals for the limit switches, some don’t. Yours doesn’t but you can add two inexpensive reed switches to sense when the door is fully open and fully closed. If you are electrically savvy you can open the gdo and check terminals 15&16 to see if they provide the signal you need (see wiring diagram in your gdo’s manual). 

obstruction sensors must be ~6v to work with v2.5 ratgdo hardware. Anything higher will burn out components. I’m working on a revision (will be several months before it’s ready) which will support a wider range of input voltages.

aram535, 2023-10-21:
@PaulWieland  are the reed sensors required? I have a camera that I use for gdo status - just need the trigger and trying to full proof as the unit is 15 years old at this point and will have to be replaced soon.

PaulWieland, 2023-10-21:
The current firmware wants to know the status of the door so if you send it an open command and the door is already open the command is ignored. 

It would be trivial to expose a “toggle” mqtt command to the firmware so that the current door status is ignored.