PaulWieland/ratgdo #143: Liftmaster RSL12v

jnewland, 2023-10-20:
My Liftmaster RSL12v mentions Security 2.0+ and has the yellow learn button, but after cracking it open it isn't entirely clear how I should wire this thing up: there are no obvious grey, white, red cables like on my similar Security 2.0+ Garage Door Opener and in the wiring instructions.

The manual does mention this connectivity for an 'alternative radio receiver':

[image]

And here's the wiring diagram for that section (and the sensors near it):

[image]

There are also these two unmentioned connectors at the bottom:

[image]

Any suggestions? I'm happy to poke at this thing with a multimeter if we need to do some digging.

rjgrandy, 2023-10-24:
In a similar vein, I have a CSL24U, similar to this and would love to have this project work with it! Would also be open to debugging.

PaulWieland, 2023-10-24:
Check to see if the edge switches are pulled to ground when the gate is fully open / fully closed.

We also need to know what that single button terminal does. I hope that it toggles the gate when connected. Whats the voltage between ctrl power and single button? How about open to ctrl power?

Nothing there mentions a serial or data connection, so I don't think we can use Security + 1/2 protocol for it...

YBonline, 2023-10-24:
> In a similar vein, I have a CSL24U, similar to this and would love to have this project work with it! Would also be open to debugging.

The CSL24U includes a datalink connection to connect it with another gate (not sure if the RSL12v does, your pictures make it look like it doesn't). I am waiting to receive my ratgdo still, but as mentioned in #90 there is a 2 pin terminal block above the "lock" position that is supposed to synchronize all of the settings to another matching control board over an encrypted data line (sounds like Security+ 2.0 to me???). I was going to test it on one of mine once my ratgdo comes in but if you want to test it... that might give you some results? You should have a green terminal block there if you have the chips, the one in the picture in 80 is missing the block if your board wasn't equipped with that option (my understand is all CSL series boards have that)

jnewland, 2023-11-02:
Aight it was finally dry and warm enough for me to do some tests.

> Check to see if the edge switches are pulled to ground when the gate is fully open / fully closed.

Bad news here on-board edge switches. Testing seemed to show them with continuity regardless of the state of the gate. So I did some research; here's some text full of typos crudely extracted from [this manual](https://support.chamberlaingroup.com/s/article/GATE-LiftMaster-CSL24U-Slide-Gate-Operator-Installation-Manual-1484145708732) with "AI" I guess lol. Looks like these are connectors for entrapment protection devices:

> CLOSE EDGE 2( Terminals)
> The CLOSE EDGE input si for edge sensor entrapment protection for the close direction. When an obstruction si sensed during gate closing the gate wil reverse ot the ful open position, disengaging the Timer-to-Close. Thsi input wilebdisregarded during gate opening.

> OPEN EYESE/DGE 2( Terminals)
> The OPEN EYESE/DGE input si for photoelectric sensor or edge sensor entrapment protection ofr hte open direction. When na obstruction is sensed during gate opening the gate wil reverse for 4seconds then stop. Thsi input wil be disregarded during gate closing.

These don't sound useful for us as edge sensors.

> We also need to know what that single button terminal does. I hope that it toggles the gate when connected. Whats the voltage between ctrl power and single button? How about open to ctrl power?

Also after closer inspection it doesn't look like I have a SBC connector on my board even though the manual for my opener says I should. My board is a K1D8059-1CC. The expansion board - K1D8387-1CC or similar - seems to have the SBC. I have no idea why this was included in the manual for the RSV12V I found. However, I don't think the SBC is what we want either. More text lifted from that PDF seems to indicate that it allows a single momentary switch to be used to toggle through a command sequence:

> Single Buton Control, SBC 2( terminals)
> Gate command sequence, Open, Stop, Close, Stop ...

However I *do* have a "link" A/B pair. Here's what I found about this in the docs:

[image]

[image]

Voltage between A and B fluctuated a bit during an open and a close. Here's a full video of a close to demonstrate it:

https://github.com/PaulWieland/ratgdo/assets/47/9a53033b-e707-4fcf-bd58-ae4bcf0457c8

That looks ... interesting? Let me know if you'd like to test it further in any other situation.

If that link port doesn't bear any fruit, page 31 of [this manual](https://support.chamberlaingroup.com/s/article/GATE-LiftMaster-CSL24U-Slide-Gate-Operator-Installation-Manual-1484145708732) does seem to indicate like the expansion board might be viable: it has two relays - AUX RELAY 1 AND 2 - that can be configured as open limit and close limit switches. Page 34 also seems to talk about open, close, and stop inputs. I'm open to trying to source one of these and do some tests on it if it sounds like we need to.

YBonline, 2023-11-04:
Did you have a ratgdo? My plan when I get one was to hook up to com link data A and B and see what happens, I'm assuming thats safe?

jnewland, 2023-11-13:
> Did you have a ratgdo?

I do, but only one of them and I'm not entirely sure the voltages this thing is emitting are in spec for the 2.0 boards I ended up with so I haven't hooked it up yet.

> Voltage between A and B fluctuated a bit during an open and a close

@PaulWieland  does the voltage on the com link in the video above ☝️ look like security 2.0 comms that would be compatible with ratgdo or are we out of luck with that angle? If we're out of luck there I'm interested in investigating adding a K1D8080-1CC and wiring my device up with its limit switches; is that supported with 2.0 boards?

PaulWieland, 2023-11-15:
that 12v signal is worth investigating. You can connect that to ratgdo's red ctrl terminal and see if its a security + 1 or 2 data line.