PaulWieland/ratgdo #145: Will ratgdo V2.5 work with Genie 3024?

rkkhanna, 2023-10-22:
I have two Genie 3024 GDOs. They don't seem to be Security 2.0 units. Shorting the two wires at wall console operates the garage door. Wiring diagram for the unit has terminals 1 and 2 for obstruction sensors (Safe T-Beam) while the wall console is wired to terminals 3 and 4. I do not see dedicated terminals for limit switches:

[image]

I ordered two new V2.5 ratgdo install kits yesterday hoping (and praying) they will allow me to get away from MyQ forever. In addition to the [wiring diagram for V2.5](https://user-images.githubusercontent.com/4663918/276749741-fe82ea10-e8f4-41d6-872f-55eec88d2aab.png), also came across Paul's [cliff notes](https://github.com/PaulWieland/ratgdo/issues/141#issuecomment-1773793929) for the new release:

> "door can be triggered
> 
> Door status can be determined using two limit switches which are wired to ratgdo’s trigger open/close terminals. Some gdos provide external terminals for the limit switches, some don’t. Yours doesn’t but you can add two inexpensive reed switches to sense when the door is fully open and fully closed. If you are electrically savvy you can open the gdo and check terminals 15&16 to see if they provide the signal you need (see wiring diagram in your gdo’s manual).
> 
> obstruction sensors must be ~6v to work with v2.5 ratgdo hardware. Anything higher will burn out components. I’m working on a revision (will be several months before it’s ready) which will support a wider range of input voltages."

Looking at the wiring diagrams for Genie and ratgdo V2.5, I need some help with wiring the two:

1. There are two "Red Ctrl" terminals on ratgdo V2.5. Do I wire both to terminal 4 on Genie? The diagram shows one Red Ctrl going to gdo unit and the other to "Wall mounted button". Do I actually need to take a wire from the top Red Ctrl terminal all the way to my wall console?
2. Does "Wht GND" get wired to terminal 3 on the Genie?
3. Likewise, there are two "Blk Obst" terminals on V2.5. Do I wire both to terminal 1 on the Genie, or one of them to terminal 1 and the other all the way to the two obstruct units on either side of the garage door? I understand I must ensure that the obstruct sensors are ~6VDC. I have a multimeter on its way to check (I am not an electrical engineer hence don't have one at home 😊)
4. I also understand I need [additional reed sensors to tell ratgdo "open" and "close" door status](https://github.com/PaulWieland/ratgdo/issues/112#issuecomment-1756718992). With these sensors, ratgdo will be able to report door states of `closed, opening, open, closing`. Do I take one of the wires from `open` reed sensor to "trigger Open" terminal and the other to "GND" terminal? Likewise `closed` reed sensor to terminals "trigger Close" and "GND"?
5. Will this reed switch (or something like this) work for the above purpose - [Rolling Door Contact Magnetic Reed Switch Alarm Sensor with 2 Wires](https://www.amazon.com/Aoje-Link-Rolling-Contact-Magnetic-Switch/dp/B08ZDDCCTR/ref=sr_1_7?keywords=garage%2Bdoor%2Breed%2Bsensor&sr=8-7&th=1)?

[image]

Thank you in advance for your time and help!

PaulWieland, 2023-10-24:
I need to acquire one of these openers and see whats on that intelligent wall console terminal...

But in the meantime, you are on the right track. 

The ctrl terminals on ratgdo v2.5 are common with each other. It just makes it easier to connect ratgdo to the opener and the wall button to ratgdo without having to splice things. One of your wall control wires (pins 3&4) is ground and the other will have + voltage. Wire the voltage line to one of the ctrl/red terminals and the other line to the ground terminal.

Next you need two of those reed switches. Mount the magnet to the top edge of the door. Mount one switch on the track so the door triggers it when its fully open. Mount the other switch on the track so that the door magnet triggers it when its fully closed. One wire from each read switch goes to ground. The other wires go to the trigger open & trigger close terminals.

Lastly is the obstruction sensor. You have to check the voltage and make sure that it is in the 5-7v range. If it is, you can connect it to the obstruction terminals on ratgdo. If its a higher voltage its not supported.

rkkhanna, 2023-10-25:
Thank you, @PaulWieland.

I understand your directions above on wiring ctrl terminals and reed switches. Will test when I get the V2.5 units. I saw your [updated V2.5 Wiring Diagram for Dry Contract Openers](https://user-images.githubusercontent.com/4663918/277820261-14136b0c-200d-479d-a55e-6097e33f4a3c.png). I assume this is what I should follow for this specific GDO.

Follow-up questions regarding obstruction sensors:

1. What functionality do I lose by not connecting them?
2. I checked the voltages on these sensors. They measure ~8.8vdc when clear and ~11.3vdc when obstructed. I came across [your post](https://github.com/PaulWieland/ratgdo/issues/112#issuecomment-1773811908) a few days back where you recommended using voltage divider to drop the signal voltage to acceptable value:

> You would need a simple divider circuit to drop the obst signal voltage.
> 
> A 10kohm x 10kohm voltage divider should work since you know you have to halve the voltage.

I am, unfortunately not an electrical engineer by trade, hence having a hard time finding this component on Amazon. When I search for it, I get a bunch of listings for 10kohm resistors [like this one](https://www.amazon.com/Resistor-Tolerance-Resistors-Limiting-Certificated/dp/B08QRJZ82J/ref=sr_1_1_sspa?keywords=10k%2Bohm%2Bx%2B10k%2Bohm%2Bvoltage%2Bdivider&sr=8-1-spons&sp_csd=d2lkZ2V0TmFtZT1zcF9hdGY&th=1). Not sure if this is the right component and how to use it if it is. Can I request you to give an example of one that would work and wiring instructions for the same?

[image]

I also noted in your cliff notes you are working on a V2.5 variant that will operate on wider range of obstruction sensor voltages, but it could take time. While I am fine with ordering new boards when you release them, I would like to use the obstruction sensor terminals with the two V2.5s that I have ordered, if possible.

Thank you for your patience and assistance.

PaulWieland, 2023-10-25:
> What functionality do I lose by not connecting them?

ratgdo won't report obstruction sensing, but control over the door and the door status feed back will work just fine.

> Not sure if this is the right component and how to use it if it is.

Yeah if you look up what a voltage divider circuit looks like you would have to take two resistors and connect them in such a way that the max voltage falls within the 5-7v range that ratgdo supports. There is no guaranty it will work though because we don't know exactly what the signal from your GDO's obstruction sensors looks like yet.  ratgdo expects there to be pulses every few ms to indicate that there is no obstruction. Lack of pulses means theres an obstruction. My test rigs all use this behavior, but yours could be different.

CrazyCoder, 2023-10-26:
See https://github.com/PaulWieland/ratgdo/issues/148#issuecomment-1780244980 regarding my Genie 8600IC findings. Yours may be the same.

rkkhanna, 2023-10-27:
Thank you, @PaulWieland and @CrazyCoder. Waiting to receive ratgdo units to work on them basis your inputs above.

coder-desai, 2023-11-10:
> ratgdo won't report obstruction sensing, but control over the door and the door status feed back will work just fine.
>
>
> Yeah if you look up what a [voltage divider circuit looks [...]

@PaulWieland  So Dry contact for GENIE 3024 will only work with a reed switch installed? Plugging in only door contact switch will not operate garage door right? Just confirming.

PaulWieland, 2023-11-10:
the mqtt ratgdo firmware wants to know the door state so if you issue it two open commands, it will ignore the second one. There is a toggle command, but without the limit switches installed you really have no idea whats going on with the door.

PaulWieland, 2023-11-10:
the esp home firmware offers a toggle command, which I might also add to the mqtt firmware when i have time

CrazyCoder, 2023-11-10:
For the basic dry contact operation without knowing the door state, any $5-$10 smart relay will work just fine (for toggling). You can add wireless door sensors and write a simple automation in HA to open/close the door and report the states. The whole purpose of ratgdo is to use the existing limit switches and keep everything wired for instant response and less maintenance.

coder-desai, 2023-11-10:
> the esp home firmware offers a toggle command, which I might also add to the mqtt firmware when i have time

My reed switches are on the way... I can wait one more day and finally say RIP to myQ :)

CrazyCoder, 2023-11-10:
@coder-desai Just curious, doesn't your current GDO setup have any open/close limit switches already connected? Why would you need additional switches?

coder-desai, 2023-11-10:
[image]

I have only two contacts accessible one is obstruction which is running on 13/8 volts and other is switch i guess. No other contacts available.

CrazyCoder, 2023-11-10:
I also didn't have the contacts accessible, but there were wires coming from the switches into the GDO. It's pretty easy to split them with wago nuts like this:

[image]

coder-desai, 2023-11-10:
@CrazyCoder  I have wago nuts handy, How to re-wire the ratgdo using the wago and splitting the wires ? sample sketch on paper will also help. You guys are very supportive and prompt providing the solution and i really appreciate that.

PaulWieland, 2023-11-10:
Heres the diagram:

[image]

My test Genie has screw terminals (without terminals you need to do what @CrazyCoder did):

[image]

[image]

coder-desai, 2023-11-14:
I have and issue with purple light blinking on my GENIE 3024 soon I wire the ratgdo to the garage door. Also wall button  led goes OFF once wirings are connected. Anything wrong i am doing here ?