PaulWieland/ratgdo #147: Backorders and v2.5

kramttocs, 2023-10-22:
Will backorders be shipped as v2.5?
If not and they will use v2.0 "old stock", can we request the 2.5 or cancel/reorder?

PaulWieland, 2023-10-22:
All remaining back orders will receive v2.5

closed 2023-10-22 by PaulWieland

MCengiz16, 2023-10-22:
Is there a timeline for backorders?

kramttocs, 2023-10-22:
Thanks @PaulWieland . Much appreciated

foreverimagining, 2023-10-23:
Since this changes a lot of things, I have a few additional questions for clarification.
- The v2.5 boards do not seem to have a shield-only option, so does that mean that the ESP8266 D1 clone will come installed on all of them, or is it a slightly different microprocessor?
- If we ordered the shield-only v2.0 board, will we still need to source our own ESP8266 D1 clones, or will the v2.5 being sent instead be the completely assembled controller?
- The wiring is a little different, which isn't necessarily a problem, but I can't tell from any of the pictures that I have found whether the v2.5 still has the screw terminals, or will we have to have a different style of connector or even do soldering at the board end?
- Is the v2.5 board different in dimensions from the v2.0 board so that the 3D-printed enclosure from this issue: https://github.com/PaulWieland/ratgdo/issues/103#issue-1925233331 will not fit properly, and if so, what are the dimensions so I can go ahead and start modeling a new one in anticipation?

samwiseg0, 2023-10-24:
@foreverimagining I plan on supporting the 2.5 version. Once I have the dimensions it will be trivial to update the model I have already created. 

In the future it would ideal if @PaulWieland could provide dimensions of the changes so I do not have to rely on others and/or my own money to keep up with the changes. 

I have tried to reach out to Paul several times and I have not heard back.

kramttocs, 2023-10-24:
@samwiseg0  Appreciate you making that enclosure. Printed one to have ready to go that completed a couple hours before I learned about 2.5 :)

foreverimagining, 2023-10-25:
> I plan on supporting the 2.5 version. Once I have the dimensions it will be trivial to update the model I have already created.

That's great to hear! Thank you so much.

While I have you, would you be able to tell me the tolerance you modeled it with? I need a gap of 0.20mm on each side of the interior of a multi-part model to achieve a snug-but-removable fit, basically a 0.20mm Horizontal Expansion setting in Cura, but I hate having to print multiple copies of models (or worse, doing a lot of sanding) if they don't happen to have the same tolerance as my printer/filament requires. If I know ahead of time, I can adapt for it in my slicer for most models, but I have at times just given up and modeled copies for myself. Alternatively, a small test print to check fit would be much appreciated.

I'd be happy to help if I get my v2.5 before you get the dimensions, but I have a feeling I'm quite a ways down the list.

> Printed one to have ready to go that completed a couple hours before I learned about 2.5 :)

And this is why I've been procrastinating getting my ESP8266s and printing the enclosure. XD I wasn't actually expecting it, and I don't particularly appreciate the lack of forewarning, but I wanted to at least have the shield on the way before I went all in.

samwiseg0, 2023-10-25:
> While I have you, would you be able to tell me the tolerance you modeled it with? I need a gap of 0.20mm on each side of the interior of a multi-part model to achieve a snug-but-removable fit, [...]

The minimum gap distance is `.245` In some places on the lid it is larger. I tried to make the gaps reasonable so it does not have to be a perfect print to work. 

> I'd be happy to help if I get my v2.5 before you get the dimensions, but I have a feeling I'm quite a ways down the list.

Thanks. That would be great. I maybe down the list as well.

foreverimagining, 2023-10-26:
> The minimum gap distance is `.245` In some places on the lid it is larger.

I think that'll work just fine, especially since I took a better look and realized it's a snap-fit lid rather than press-fit. Now, it's just a matter of time and updated dimensions.

PaulWieland, 2023-10-26:
Sorry for the lack of response, have been rather... busy...

I'll get you guys some outline drawings as soon as I can. I'll open a separate repo up on github.com/ratgdo for 3d printed stuff.

kramttocs, 2023-11-07:
My two units show out for delivery today. @samwiseg0 I can provide dimensions and test some prints if yours or @foreverimagining  haven't shipped yet.

samwiseg0, 2023-11-07:
> My two units show out for delivery today. @samwiseg0 I can provide dimensions and test some prints if yours or @foreverimagining  haven't shipped yet.

@kramttocs I _think_ I have everything I need for v2.5. I just need someone to print and test before I release it. 

Can you reach out to me via email and I can send you the beta files?

3d@watch-tower.net

foreverimagining, 2023-11-07:
I won't be much help for a few more days myself since I only finally got my tracking number last night. I figure the model will be ready to go by then, but I'll keep my eye out.

samwiseg0, 2023-11-07:
> I won't be much help for a few more days myself since I only finally got my tracking number last night. I figure the model will be ready to go by then, but I'll keep my eye out.

@foreverimagining No worries, it is already live. https://github.com/PaulWieland/ratgdo/issues/103

foreverimagining, 2023-11-07:
I actually checked right after I responded and saw that. XD