PaulWieland/ratgdo #148: Ratgdo Compatibility Thread

ljmerza, 2023-10-23:
Instead of having a new issue for every compatibility question we can make a mega issue for those questions. Out your model and someone can answer in a thread to your comment if it's compatible. Be sure to search for your model before asking for compatibility.

ljmerza, 2023-10-23:
Is Genie 2055 for ratgdo v2.0? I just got mine right before v2.5 was released so may have to switch.

ljmerza, 2023-10-23:
would be easier for others to respond here and update there accordingly

PaulWieland, 2023-10-23:
Yeah. As people respond it can be added to the wiki page for others to reference instead of having to read through a thread.

barracksbuilder, 2023-10-24:
I have a craftsman 1/2hp Model 139.53964SRT (Date: 10/2002) The control board side has a part number 41A5021-4F. Craftsman are rebadged Chamberlain operners. This unit only has 3 terminals exposed (1 Red and 2 white). I am able to short the 1 red and 1 white to open/close the door. From my reading, it is a Security 1.0 device (Red/Orange learn button) Would the ratgdo 2.5 work? If so what would I need to wire up since it only has 3 screw posts?

PaulWieland, 2023-10-24:
I'm assuming they just have 1 terminal for ground instead of 2, but can you post a picture? Where are the obstruction sensors connected?

barracksbuilder, 2023-10-24:
[image]

My door control button is in the red #1 screw and white #2 screw, the safety beams are black #3, and white #2

Kaldek, 2023-10-24:
I have an MJ3800MYQ door opener marked in all documentation as Security+ 2.0 but cannot tell if it is supported.
The wiring is weird.  Pins 1 and 3 are for the "E-Serial" Smart Control Panel 880AUW.  Pin 2 is a dry contact for installing your own button which is odd because the "red wire" is the control wire on all other examples I have seen using ratgdo.

FYI, this is an Australian unit.

[image]

[image]

PaulWieland, 2023-10-24:
> [image]
> 
> My door control button is in the red #1 screw and white #2 screw, the safety beams are black #3, and white #2.
> 
> Next to the black #3 screw it has a 24v sticker (not sure why it's there). But tested when not obstructed it's 5.88v when there is an obstruction it is 6.24v

You have all three terminals that are needed. White is ground. The newer openers provide two separate ground terminals to avoid having two wires twisted together. 

So wire ratgdo accordingly. red/ctrl -> red, white/gnd -> white, black/obs -> black

PaulWieland, 2023-10-24:
> I have an MJ3800MYQ door opener marked in all documentation as Security+ 2.0 but cannot tell if it is supported. The wiring is weird. Pins 1 and 3 are for the "E-Serial" Smart Control Panel 880AUW. [...]

This is what I'm calling a "ROW" opener - it seems that most openers outside North America have the same terminal setup as yours. Connect terminal 1 to red/ctrl on ratgdo. I need to add this diagram to my wiring instructions page.

STDEin, 2023-10-25:
I have a Skylink garage door opened (rebranded as Canadian Tire Mastercraft AT-803) that according to the Chamberlain [List of compatible garage door openers for MyQ Garage](https://support.chamberlaingroup.com/s/article/List-of-compatible-garage-door-openers-for-MyQ-Garage-1484145635905) page has partial support "Only able to monitor the status of this Garage Door Opener"

Would that extend to the ratgo? Monitoring but no control, or is it possible that it could control the opener completely?
The opener does not respond to shorted wires to the control pad, but only the the button press on the panel itself.

Kaldek, 2023-10-25:
> This is what I'm calling a "ROW" opener - it seems that most openers outside North America have the same terminal setup as yours. Connect terminal 1 to red/ctrl on ratgdo. I need to add this diagram [...]

Fixed and working.  I just had my circuit design wrong.

CrazyCoder, 2023-10-26:
Genie 8600IC: https://www.geniecompany.com/files/en/manual/is-isl-ic-h-series.

Dry contact.

Voltage between wall controls terminals (button): 5.8V
Voltage between the open/close switch terminals: 7.85V
Voltage between obstruction sensors terminals: 12.5V

Open/close switch wires/terminals are not exposed. Ground is the screw rail itself. 2 wires from open/close switch contacts go inside the box and need to be spliced for connection. Voltage divider would be needed to be compatible with the current board versions.

[image]

Signal is HIGH when there is an obstruction.
Signal sends PULSE when there is no obstruction.

The pulse looks like this:

[image]

The low pulse length is between 0.4 and 0.5ms.

Will this work with the current pulse detection code if the proper voltage divider is added? What would be the best resistor values for such a divider?

mjmcg, 2023-10-27:
Hi @PaulWieland - Love this project. I ordered 1 for my Chamberlain, and am wondering if ratgdo will work with my Legacy Overhead door. Here are pictures of it along with the wiring diagram:
[image]

[image]

[image]

PaulWieland, 2023-10-27:
@mjmcg That age chamberlain opener is uncharted territory, we have to do some investigation to see if it has a serial interface or if its just dry contacts.

What wall control do you have? Is it just a door bell or is it multi function?

Worst case scenario you will be able to control it with the dry contact interface. You will have to add two reed switches to act as limit switches for fully open and fully closed.

mjmcg, 2023-10-27:
Thank you for the quick reply! Here's are pics of the remote, along with the circuit board. It has a light and an open/close button. Happy to add the reed switches. Where would I add them?

[image]

[image]

[image]

PaulWieland, 2023-10-27:
I need to do a write up for this because its going to be a FAQ. 

Get two reed switches. Mount the magnet to the top edge of the door. Mount the "open limit" reed switch on the track so that the switch is closed when the door is fully open. Mount the other reed switch farther down the track so that it is closed when the door is fully closed.

One wire on each reed switch is connected to Ground. The other wires need to be connected to ratgdo terminals Trigger Open & Trigger Close respectively. [See the Genie example on the wiring page.](https://paulwieland.github.io/ratgdo/03_wiring.html)

bdb12, 2023-10-28:
> I have an MJ3800MYQ door opener marked in all documentation as Security+ 2.0 but cannot tell if it is supported. The wiring is weird. Pins 1 and 3 are for the "E-Serial" Smart Control Panel 880AUW. [...]

Hi [Kaldek](https://github.com/Kaldek)

Is there a chance you could post some pics of your install including the Ratgdo unit? I have the same opener as you and just wanted to see what you did. Can I also assume you used a power brick for USB vs finding something on the opener to power the Ratgdo?

Thanks in advance

B

PaulWieland, 2023-10-28:
@bdb12 the wiring is very straight forward. 

ratgdo - opener
Red/ctrl - 1 eserial 
Wht/gnd - 3 or 4 ground
Blk/obst - 5 ir sensor

the only difference between the 14 terminal ROW openers and NA openers in terms of wiring is the red terminal (it’s green for you). Our openers in NA don’t have a dry contact input.

bdb12, 2023-10-28:
> @bdb12 the wiring is very straight forward.
>
> ratgdo - opener Red/ctrl - 1 eserial Wht/gnd - 3 or 4 ground Blk/obst - 5 ir sensor
>
> the only difference between the 14 terminal ROW openers and NA [...]

Thanks so much Paul. Just ordered a Ratdgo about an hour ago. Is the current lead time still 2-3 weeks?

Goose66, 2023-10-28:
I have a Genie CM7600IC. It has limit switches built-in, and wall control panels with separate light and door buttons, as well as a lock. Is there a secret specification item or class of Genie controller boards that we are looking for for compatibility?

PaulWieland, 2023-10-28:
@Goose66  as far as I know Genie doors can be controlled with simple dry contact. ratgdo will pull the control line to ground for 1/2 a second which is enough to operate my legacy unit. The limit switches provide feedback so ratgdo knows the four door states.

I'm going to buy a newer genie opener to see what their interface looks like since I might be able to add support for light and lock controls.

Goose66, 2023-10-28:
I don't think mine is actually "newer," but I would be happy to help. Just need to get my hands on an oscilloscope (and someone who knows how to use it).

amferguson, 2023-11-03:
I apologize if this is not the correct location to ask this, but does anyone know if this would work for Linear models (labeled as Precision PDS-800)? It does work with my myQ-G0401-ES if that helps at all. I'm not sure how to get a wiring diagram, but here is what the connections coming down from the ceiling look like. 
[image]

Basically in a similar boat to this comment (https://github.com/PaulWieland/ratgdo/issues/159#issuecomment-1787630747) and am trying to flee the myQ ecosystem since it keeps having issues with HomeAssistant. Thanks!

richardpowellus, 2023-11-03:
[image]

I think this is a model 8522 which I believe is supported, but I'd love confirmation before I order two kits...

PaulWieland, 2023-11-04:
> [image]
> 
> I think this is a model 8522 which I believe is supported, but I'd love confirmation before I order two kits...

Yes its supported. Thats a security + 2.0 (yellow learn button) opener.

haffi78, 2023-11-06:
I just orderd mine, this is how my Chaimberlain looks like, its an EU model. I assume this will just work i.e. the button today is 1 and 2, and door sensor 2 and 3.
Or is the diagram different for this type ? 
[image]
 
Thanks !

dcode, 2023-11-06:
I've got two 8500W openers, which are the wall mounted jack openers with WiFi and battery builtin. These openers have the usual beam obstruction sensor as well as some sort of reed switch up top, never really figured out what that was for (one door is missing the paddle so clearly useless).

It also has a lock (looks like headers for possibly 2 locks) on the side of the channel and wirelessly pairs with a light.

I feel like these were the upgraded from 8500C and use Security+ 2.0, but sounds like the v2.5 boards will work even if it's 1.0.

If I get it all wired up correctly, does the board report the status of the lights and lock as well?

[image]

[image]

[image]

PaulWieland, 2023-11-06:
> If I get it all wired up correctly, does the board report the status of the lights and lock as well?

It reports the status of the wireless lockout. If the mechanical lock status is reported on the serial line we can report that too, but it would take some debugging from you once you have a ratgdo installed so that we can identify if the lock status is communicated and if so, what the data looks like.

PaulWieland, 2023-11-06:
> I just orderd mine, this is how my Chaimberlain looks like, its an EU model. I assume this will just work i.e. the button today is 1 and 2, and door sensor 2 and 3. Or is the diagram different for this type ? 

The ROW openers have a different terminal for the serial data. Your manual should show one of them as being "eserial" - usually its the first terminal which is sometimes labeled `#0` and sometimes `#1`. See https://github.com/PaulWieland/ratgdo/wiki where I mention the openers with 14 terminals.

MikeMcGlinch, 2023-11-09:
I have 2 Chamberlain GDOs.  Do I need 1 of your devices for each Opener?  Or will one device operate both?

rlowens, 2023-11-09:
> I have 2 Chamberlain GDOs. Do I need 1 of your devices for each Opener? Or will one device operate both?

You need one for each door. @PaulWieland maybe add this to the FAQ? I've seen this question at least 3 times in various places.

larryopoly, 2023-11-09:
> You have all three terminals that are needed. White is ground. The newer openers provide two separate ground terminals to avoid having two wires twisted together.
>
> So wire ratgdo accordingly. [...]

@barracksbuilder, I seem have the same model, just a year older (41a5021-3E). Mind reporting back if/when you're up and running?

thickconfusion, 2023-11-11:
Compatibility with Genie 2028?

tigeinc, 2023-11-11:
Hello, have purchased a 2.5 board, hoping sound out compatibility with a Merlin SilentDrive Elite MR865MYQ.
This GDO and its variants are currently sold and seem to be fairly common in Australia and New Zealand.
[MR865MYQ manual](https://p.widencdn.net/ynlory/MR865MYQSilentDrive-Elite-MyQ-Manual-114A4903C)
[image]
[image]
[image]
[image]
It uses Security2.0+, but has a different learning/program button arrangement.
No explicit reference to eserial in the manual or documentation.

Wondering if anyone has installed with this unit or encountered similar?

NukeThemTillTheyGlow, 2023-11-13:
Before ordering what appears to be 3 v2.5 boards, I thought I would confirm the details of the 3 openers I have:
**Manufacture	Model	                        Model number   Button**
Liftmaster	Professional Formula 1	2280-267	          Red
Liftmaster	8550W	                        8550W	          Yellow
Liftmaster		                                87504-267	  Yellow

shanelord01, 2023-11-14:
**Working - Australian Merlin Door Opener**

Model: Merlin MS105MYQ
Country: Australia
ratgdo version: 2.5i
s/w: ESPHome

Connectors:
Merlin 0 - eserial (red/Ctrl)
Merlin 2 - ground (white/GND)

pixelrazor, 2023-11-15:
Sharing my experience hooking up my Genie 1028:

Hooking the voltage divider to the other side of the obstruction sensor was putting out a lower voltage than hooking it to the other ground. It shouldn't make a difference but it did. Could also be due to bad contact with the twisted wires. Either way, the obstruction sensing seems a bit flaky - I'll try to collect more usage to see. I have reed switches on the way for the door state, but opening and closing works fine. The voltage divider is with two 10k resistors.

[image]

PaulWieland, 2023-11-15:
I need to write an auto sensing frequency calculator for the obstruction input. Not all doors use the same frequency. I think some (if not most) genie openers use a much slower pulse than the chamberlain ones. Sharing my experience hooking up my Genie 1028:
Hooking the voltage divider to the other side of the obstruction sensor was putting out a lower voltage than hooking it to the other ground. It shouldn't make a difference but it did. Could also be due to bad contact with the twisted wires. Either way, the obstruction sensing seems a bit flaky - I'll try to collect more usage to see. I have reed switches on the way for the door state, but opening and closing works fine. The voltage divider is with two 10k resistors.

CrazyCoder, 2023-11-15:
> I think some (if not most) genie openers use a much slower pulse than the chamberlain ones

At least my model has faster pulses than ratgdo is expecting: https://github.com/PaulWieland/ratgdo/issues/168.

MRobi1, 2023-11-16:
My dad's got an old Liftmaster opener. I'm unsure of the model number at the moment, but it's got a green learn button. It's definitely a dry contact opener so I know I could get the open/close to work. I'm just wondering if it would be able to read the door position off the motor like it does with the newer openers