PaulWieland/ratgdo #149: Web Installer Help

raf63086, 2023-10-24:
I am having trouble using the ratgdo esp web installer.  My computer seems to recognize there is a device and I have confirmed I am using a data cable and not a power only USB cable.   When I click 'Connect' on the web installer I am able to select the port, go through the menus to install ratgdo, and confirm installation, however the installation fails with the error:

'Failed to initialize.  Try resetting your device or holding the BOOT button while clicking INSTALL.'

I have a v2.0 board.  Tried using both Microsoft Edge and Google Chrome.  No lights come up on the board, I'm not sure if it is supposed to.  Also attempted to use the ESPHome port installer and got a similar error.  Also tried to install on a Mac laptop but got the same error there as well.

Any troubleshooting suggestions would be appreciated.

[image]

PaulWieland, 2023-10-24:
whats your email address or order number?

raf63086, 2023-10-24:
I believe it is 11ee5e5687b950189b78ac1f6bbbcc9c

PaulWieland, 2023-10-24:
Send an email to ratgdo@gmail.com

closed 2023-10-25 by PaulWieland