PaulWieland/ratgdo #154: Possible to control connected gate opener?

TopBanana123, 2023-10-25:
I have a MyQ Opener which is connected to WiFi, and my LA400 gate opener is connected to the garage door opener, which then lets me control both via the Garage Door Opener/App

Is there any possibility for the Ratgdo to also control the already connected gate opener?

PaulWieland, 2023-10-25:
No confirmations yet, but there are several already open issues about interfacing with gate openers. It seems as though there is a serial interface on at least some of the gate controllers. Others seem to have limit switches and dry contact control which would also work (just requires an extra wire).

TopBanana123, 2023-11-01:
> No confirmations yet, but there are several already open issues about interfacing with gate openers. It seems as though there is a serial interface on at least some of the gate controllers. Others [...]

Thanks. If using the dry contact control or SBC, I assume we would have to figure out another way to have the gate open status?

PaulWieland, 2023-11-01:
Duplicate #90

closed 2023-11-01 by PaulWieland