PaulWieland/ratgdo #155: DIY Shield

rohinrohin, 2023-10-25:
Hey @PaulWieland! Great project, thanks a ton for this. The recent MyQ changes is what got me here. 

I need this integrated super urgently and can't wait for the shields to arrive. 

Is there a schematic/shield pin mapping diagram for me to DIY the shield on my own?

In the spirit of open source this would really help me and others out too! 

Happy to contrib back on instructions on how to set this up without the shield if that's something that your cool with.  

Thanks,
Rohin

Kaldek, 2023-10-25:
Paul is refusing to provide this so I reverse engineered it:
https://github.com/Kaldek/rat-ratgdo

rohinrohin, 2023-10-25:
Awesome @Kaldek! 👏 thanks for this :)

Kaldek, 2023-10-25:
> Awesome @Kaldek! 👏 thanks for this :)

I've done some updates to my repo with a recommendation for a different MOSFET on the TX line.  Just in case you had already committed to parts.  No big deal, it should work regardless but may not have 100% reliability of TX.