PaulWieland/ratgdo #156: Flashing issue with web installer

optik678, 2023-10-25:
Hello, having an issue flashing my microUSB D1 mini boards.  I purchased two shields and then a whole slew of USB-C and micro USB D1 mini boards from overseas. The first D1 Mini flashed ok (ESPHome version) and is installed and operating now. I have tried on two other micro USB boards and I get the same errors on both boards. 

----
For ESPHome firmware (i select 8266, The firmware seems to flash but I can't connect to WiFi or get an IP address.

There seems to be some strange mdns errors. The log just constantly repeats these error lines until the log bugger fills on the tool. How else can i troubleshoot this? Do you recommend trying a 3rd board?

Text that spams the console/log until the device becomes unresponsive (about 30 seconds or so) :

[D][ratgdo:464]: Collision detected, waiting to send packet
[W][component:214]: Component mdns took a long time for an operation (0.05 s).
[W][component:215]: Components should block for at most 20-30ms

Tried to connect to wifi with my phone and browse to ratgdov2.local:80, 8266, and 6053 but nothing pulls up a webpage

---
For vanilla ratGDOv2 flashing, **I can install WiFi disabled version perfectly fine, that works**. If I try to install 2.1 or 2.26a, it won't complete a setup. The tool will say that It completed the flash but it doesn't come up. I've tried with and without selecting 'Erase Device'

I get this in the console log after flashing if i reset the device :

Mounting LittleFS...
Failed to open [setup.json] file
Failed to parse [setup.json] file
No setup file

-----------

I've attached log files from both flashing attempts.
[ratgdo v2.26a log.txt](https://github.com/PaulWieland/ratgdo/files/13161313/ratgdo.v2.26a.log.txt)
[esphome log.txt](https://github.com/PaulWieland/ratgdo/files/13161314/esphome.log.txt)

optik678, 2023-10-25:
All of this was done with boards not connected to the shield, if that matters. One doesn't even have pins soldered to it, it's fresh out of the bag.

Kaldek, 2023-10-25:
FYI the collision detected just means that ratgdo thinks the serial line is in use.  Ignore that entry.

Kaldek, 2023-10-25:
Also your logs say that it's spinning up its own AP as per Tasmota style (even using the 192.168.4.1 standard):
```
[C][wifi:038]: Setting up WiFi...
[C][wifi:051]: Starting WiFi...
[C][wifi:052]:   Local MAC: A4:CF:12:FC:9A:26
[C][wifi:230]: Setting up AP...
[C][wifi:232]:   AP SSID: 'ratgdov2-fc9a26'
[C][wifi:233]:   AP Password: ''
[C][wifi:242]:   IP Address: 192.168.4.1
```

Kaldek, 2023-10-25:
Do you get anything like this?
[image]

Kaldek, 2023-10-25:
OK I get that mdns error if I "skip" WiFi setup.  Don't skip WiFi setup maybe?

PaulWieland, 2023-10-25:
The collision detected message from the ESPHome attempt happens when you do not have ratgdo connected to the garage door opener. It needs to receive signals from the obstruction sensor, and when it doesn't it thinks theres an obstruction.

PaulWieland, 2023-10-25:
The no setup file message is normal for a new install. That message should disappear after you restart. If it doesn't, then there could be something wrong with the flash storage of the device.

closed 2023-10-25 by PaulWieland

optik678, 2023-10-25:
So it works if I don't skip the WiFi setup, as Kaldek suggested. I flash ESPHome on top of it and the Wifi connection was challenging. Right after flashing, it wouldn't connect. Here's what eventually worked : 

1) Refresh Page, reconnect to device
2) Logs and Console -> Reboot Device, back to flasher menu
3) Click 'Connect to Wi-Fi'
4) Select AP and try to connect (failed again)
5) Press Back
6) Try to connect to Wifi again, works that time

ESPHome 2023.10.3

optik678, 2023-10-25:
Is there a console command to change the device IP for this ESPHome fork? Or do i need to alter configuration files and flash it with a separate flashing program? 

I noted in some research that these devices are retaining the initial DHCP addresses and then not pulling new ones when reconnecting and this is also what I'm observing. I have my router setup to give this device a static DHCP address via its MAC address but the device doesn't seem to poll for a new address when connecting to WiFi.

PaulWieland, 2023-10-25:
You definitely cannot skip the wifi setup. I would also suggest doing a complete erase if switching between ESPHome and MQTT versions of the firmware.

In the MQTT firmware the IP address is changeable via the web ui setup page.