PaulWieland/ratgdo #159: Can a door MYQ sensor MYQ-G0302 be read?

dougransom, 2023-10-26:
Can this [G0302](https://www.homedepot.com/p/Chamberlain-MyQ-Garage-Door-Sensor-MYQ-G0302/303335535)  be accessed and decoded?  Not by ratgo obviously but is ther a way?

PaulWieland, 2023-10-26:
No. If you have ratgdo, you shouldn't need that sensor.

closed 2023-10-26 by PaulWieland

Lebo-Wang, 2023-10-31:
Thanks! my situation is similar. I have a [g0301 hub](https://www.amazon.com/Smart-Garage-Opener-Chamberlain-MYQ-G0301/dp/B075H7Z5L8?th=1) along with the same sensor. But the opener's brand is [Linear](https://linear-solutions.com/garage-door-openers/). I am not sure if ratgo is compatible? At least the buttons are different from the ratgo's manual. Thanks a lot

Lebo-Wang, 2023-10-31:
after carfully comparing with ['v2.0 diagram'](https://user-images.githubusercontent.com/4663918/235453980-04a642fa-a181-4297-b4f3-06e1315e02fa.png) , i think it might be same with Linear opener with some inputs: common, pushbottom, beam . can take a try