PaulWieland/ratgdo #162: Temperature sensor

towerhand, 2023-10-27:
My Chamberlain garage door wall switch has a temperature sensor, and I'd like to know if the ratgdo can also get the information from the sensor since it has access to everything else.

PaulWieland, 2023-10-27:
Based on my research and testing I do not believe that the temperature is transmitted to the garage door, so ratgdo can't read it. That wall panel just has a temp sensor built in and an LCD to display the temp, but theres no reason for the door opener to know the temp so its not communicated on the bus.

closed 2023-10-27 by PaulWieland

towerhand, 2023-10-27:
What about changing the time from the wall panel from the garage opener (ratgdo)?

PaulWieland, 2023-10-27:
Not 100% sure, but I think its the same deal that the time is only local to the wall panel.

towerhand, 2023-11-01:
Ok, thanks. What about the state of the backup battery of the garage opener?

ripnet, 2023-11-13:
@PaulWieland Any thoughts?

The backup battery status should be available somehow, as the wall display shows it, so the GDO must communicate that.