PaulWieland/ratgdo #167: Ratgdo 2.0 with 8550WLB - Light switch is inverted if motion detection wall button is used

m3ki, 2023-10-28:
I got Ratgdo 2.0 with 8550WLB wired in and everything seems to work, except that when the motion sensor in the garage wall button is triggered and the light turns on, the light switch in homeassistant status is not reflected. 
What happens then is that if I press the light button in the homeassistant it turns off the physical light, however the home assistant light switch turns on.

STR

1. make sure garage door is closed
2. make sure the physical light is off
3. walk by the physical door button 
4. ** notice light button in Hass is still off
5. press the light button in HASS
6. ** notice physical light tuns off, but hass light button turns on.
7. Press hass light button
8. ** notice physical light turns on, but the hass button turns off.

PaulWieland, 2023-10-28:
Yeah I think I understand what's happening there. Might be an easy fix if I issue a query command shortly after motion is detected.

m3ki, 2023-10-28:
Does ratgdo receive motion events from door wall buttons?