PaulWieland/ratgdo #168: Obstruction is not always cleared with Genie GDO

CrazyCoder, 2023-10-28:
This condition doesn't always work with Genie since it has a higher impulse count within 50ms:

```
		// check to see if we got between 3 and 8 low pulses on the line
		if(obstructionLowCount >= 3 && obstructionLowCount <= 8)
			obstructionState = 1;
```

[image]

With the ~326Hz frequency, there will be ~16 pulses in 50ms.

Either adjust the condition or make the count configurable to be compatible with Genie openers.

My findings are for this model: https://github.com/PaulWieland/ratgdo/issues/148#issuecomment-1780244980.

I'm using a custom voltage divider.

CrazyCoder, 2023-10-29:
Changed the condition to `obstructionLowCount <= 20` and it started to work reliably.

PaulWieland, 2023-11-02:
Will be addressed in next release.