PaulWieland/ratgdo #172: I messed up: got wrong hardware version for "purple button" opener

rickestrada, 2023-10-30:
Hi! Was super excited to setup my ratgdo, but after flashing, I realized something was wrong when upon applying power, my garage door opened on its own... anyway fast forward after some reading, I realized I have a "purple" button opener and I have a v2.0 board. Can you give me some direction on how to order a replacement? Thanks!

PaulWieland, 2023-10-30:
Hi @rickestrada I'm sorry about that, but yeah the v2.0 hardware was for Security + 2.0 openers. With version 2.5 I have expanded it to work with Security + 1.0 as well as dry contact openers. All new orders will receive v2.5. v2.0 is no longer in production.

closed 2023-10-30 by PaulWieland

rickestrada, 2023-10-31:
Hi! Thanks for your reply! OK good deal, I'll order a 2.5... do I need the ESP or just the board? Actually, I don't see a way to order just the board, but wanted to confirm with you first before I buy another one (don't want to mess up again!) :) Thanks so much!