PaulWieland/ratgdo #173: Collision Detected

chernerd, 2023-10-31:
After hooking everything up and loading esphome I am receiving an error in the esphome log. HA recognizes them and they appear under the ESPHome page and in the devices/entities.

`Collision detected, waiting to send packet`

I have double and triple checked my wiring and have been unable to fix the error. Attached are pictures of the wiring and garage door pieces.

[image]
[image]
[image]
[image]
[image]
[image]

PaulWieland, 2023-10-31:
What happens with the "OEM" MQTT firmware?

chernerd, 2023-10-31:
I used a new esp board and flashed the OEM MQTT firmware. When I attached to the ratgdo I see it auto discovered in home assistant. None of the commands are working (up/down/light).

PaulWieland, 2023-10-31:
Did the door and light respond on the esphome version? If so it’s a rolling code issue. try the 2.26a release of the mqtt version

chernerd, 2023-10-31:
No, I haven't been able to get the door to respond on either version.

PaulWieland, 2023-10-31:
Is the mqtt version saying that there’s an obstruction detected? 

Double check your ground connection (white terminal). You can also try the other ground terminal on the opposite side of the board. There should be continuity between both ground terminals, the gnd pin on the d1 and both white terminals on the gdo.

chernerd, 2023-10-31:
It does say obstruction detected.

Do I have the d1 attached correctly? I'm using the connector in my picture above to attach the d1 to the gdo.

PaulWieland, 2023-10-31:
It looks correct but I can’t see for sure which wire is in which terminal. 

Do you have a multimeter that you can check continuity with? I still think the ground isn’t connected because both the obstruction and gdo control are not working.

chernerd, 2023-11-01:
What should I check with the multimeter?

With everything hooked up my garage door still works with the button.
Doesn't that mean the obstruction sensors are hooked up correctly/grounded?

I tried switching the ground to the other side of the gdo and am
still getting the same error.

rlowens, 2023-11-01:
> What should I check with the multimeter?

Confirm continuity between the white terminals on the GDO and GND on the ratgdo (GND on D1 mini and GND on the terminal blocks).

chernerd, 2023-11-02:
It doesn't have continuity. How am I supposed to be attaching the d1 to the
shield? I'm using my black pronged piece shown in my original post.

rlowens, 2023-11-02:
Ah, I see the confusion. The "black pronged piece" is called a Pin Header and you need to solder the short end of the pins into your D1 Mini board like this:
[image]

Here's someone doing it on YouTube: https://www.youtube.com/watch?v=X7iBkSICqvQ

You can't just pin the D1 Mini to the ratgdo and get a good connection:
[image]

I was going to suggest the possibility of purchasing a D1 Mini with the headers pre-installed, but 10 pages of Amazon didn't have any. Weird. The NodeMCU-style boards I use almost always come pre-soldered, but it looks like D1 Mini-style boards mostly don't.

chernerd, 2023-11-04:
Thanks! I soldered the board this morning and was able to get everything
working. It should have been obvious that I needed to solder but because it
wasn't explicitly called out I never considered it (my fault).

closed 2023-11-05 by chernerd

PaulWieland, 2023-11-05:
> Ah, I see the confusion.

Good eye. I totally missed it in the provided photo. Thank you for the help👍