PaulWieland/ratgdo #177: Trigger remote close notification (beeps/flashing) when closing

slords, 2023-11-01:
The 829LM is a remote device that has a close door button that when triggered will cause the lights to flash and unit to beep before closing.  This is the same as when triggered through the app.  Is there any way to make it so when ratgdo triggers a close for it to perform the same routine?

Stromos, 2023-11-03:
This is actually a worry of mine that could be a way for Chamberlain to go after RATGDO. Current setup is a direct violation to UL 325.

PaulWieland, 2023-11-05:
It is being worked on. [See here](https://github.com/ratgdo/esphome-ratgdo/pull/43)
We're looking for wireless command that is being sent by the hub. In theory the right value can be also sent over the wireline that will trigger the on board beeper.

slords, 2023-11-05:
I'm not sure if ratgdo does (or has the ability to) capture the commands sent from the wall button to the unit.  If it does then we might be able to determine what is sent on the wire.  My setup is a Chamberlain HD900D garage door opener which is only Security+.  I wanted to add Security+2.0 and MyQ to my system so I purchased the 889LM (MyQ Control Panel) as my wall unit which adds that capability to the system but at the wall switch.  So all the MyQ and Security+2.0 devices connect to the wall unit and then are sent to the opener over the wire.  Both my 828LM (MyQ gateway) as well as the 829LM (garage door monitor) connect to the 889LM control pannel.  Everything they do including closing the door with flashes and beeps are being sent over the wire to my unit.

PaulWieland, 2023-11-05:
I have an 889lm on my test rig and it does not make the opener beep unless you use the time to close feature. 

Are you sure yours beeps if you just press the button to toggle the door (with ttc disabled)?

rlowens, 2023-11-05:
Can we sniff the Security+ 2.0 commands from 828LM (MyQ gateway) to 889LM (MyQ Control Panel)? Also might want to sniff the Security+ 1.0 commands (889LM to GDO) when that happens so we can add those too.

PaulWieland, 2023-11-05:
ratgdo reads all data sent over the wireline. But so far I dont know of a wired setup that causes the GDO to beep which is why I asked for clarification on the 889lm setup. Wireless commands can cause it to beep but you need a SDR to capture the packets. Thats whats being worked on in the PR I linked to.

slords, 2023-11-05:
In my situation I have devices that connect to the 889LM that do cause it to beep and flash.  I have the following situations:

button on 889LM - close normal no beep or flash
normal remote to 889LM - close normal no beep or flash
keypad to 889LM - close normal no beep or flash
TTC enabled on 889LM when timeout - warning close with beeps and flashing
828LM to 889LM  signal via MyQ app - warning close with beeps and flashing
829LM to 889LM push button on 829LM - warning close with beeps and flashing

So in my situation there are two ways I can make the system close the garage door that will flash and beep.  Hopefully this helps you setup a situation where you can read what is on the wire that makes it flash and beep.  If all the flashing and beeping is being done on the 889LM and just sending signals to the opener at least you should be able to see what signals and duplicate them.

rlowens, 2023-11-05:
Does 829LM connect to 889LM via wire or wireless?

slords, 2023-11-05:
829LM (or 828LM) to 889LM is wireless but 889LM to opener is wired.

PaulWieland, 2023-11-05:
Ok then we need to capture the close command that causes the beeping when sent to the gdo from the 889lm.

slords, 2023-11-05:
As soon as I receive the unit and get it hooked up then I'll provide what I can.  Placed my order last week and it looks like they are backordered.

PaulWieland, 2023-11-06:
After thinking about it more, I bet its the 889LM itself thats beeping and not the GDO. I'm doubting the serial information sent to the GDO is any different.

slords, 2023-11-06:
I'll check and see where the beeping is coming from.  I am sure the lights flashing is being done at the GDO side.

slords, 2023-11-07:
I can confirm that the beeping is coming from the 889LM.  It might be doing all the automation and just sending flash light codes to the GDO.

CrazyCoder, 2023-11-13:
I solved this problem using a separate device: https://github.com/Kaldek/rat-ratgdo/issues/40 (as my GDO doesn't support it internally).

ozpaulb, 2023-11-13:
I bought an 819LMB MyQ Home Bridge just to hack it (I've never connected it to my gdo)

I noticed that there's a site on the PCB for a buzzer/beeper, but it's not populated.  there's also a site in the center of the PCB that appears to be for LEDs - also not populated.  I'm comparing to the PCB photo in this article: https://www.mcafee.com/blogs/other-blogs/mcafee-labs/we-be-jammin-bypassing-chamberlain-myq-garage-doors/, which appears to show both a buzzer (round object at top/center of PCB) and a big LED in the center - neither of those exist on mine.

as I've never connected it to my gdo, I don't know if it will make the gdo beep or not.  I assume it will, since this seems to be a safety feature from Chamberlain.  maybe the 889LM that you have will work with gdo's that don't have the beeper (so your device has one), but maybe mine only works with newer gdo's that are known to have a beeper.

I've extracted the wifi module's firmware, but I think the 'real' control of the gdo / wireless interface to gdo are controlled by the PIC chip.  not sure if I'll be able to extract the PIC code image - waiting for a PICkit3 device to arrive so I can try it out

ozpaulb, 2023-11-14:
oh well, I guess I can't extract the PIC firmware.

I received my PiCkit 3 device today, but when I go to read the memory of the PIC chip, it's all zeros.  the "CP0" (code protection) bit is set in the configuration, which seems to make the device read-back all zeros :( 

since the wifi module didn't have any protection enabled (88MW300 CPU), I had hoped that the PIC chip was also unlocked.  I guess not