PaulWieland/ratgdo #178: Chamberlain  opener - how to wire

bstork0807, 2023-11-02:
I understand how to wire the side with obstacles and  control. Is it necessary to wire the trigger and status side? Do I need to put reed switches on the door to show status?

PaulWieland, 2023-11-02:
Hi @bstork0807 what kind of chamberlain opener do you have? 
Is it security + 2.0, security + 1.0 or something older?

bstork0807, 2023-11-02:
chamberlain WD1000WF with security + 2.0

bstork0807, 2023-11-06:
Hey Paul   just wanted to touch base because I wanted to place an order if this will work for me.  opener is a 
chamberlain WD1000WF with security + 2.0

PaulWieland, 2023-11-06:
Its a security + 2.0 opener, so yes its supported.

See here https://github.com/PaulWieland/ratgdo/wiki
And here https://paulwieland.github.io/ratgdo/03_wiring.html

closed 2023-11-06 by PaulWieland

bstork0807, 2023-11-07:
Thanks Paul.. I guess my main question is.. Do I need to use any of connections i have marked in red?

[image]