PaulWieland/ratgdo #181: Ratgdo unable to connect to Wifi

astoltzfus, 2023-11-03:
I've had this happen many times, despite erasing and reflashing the device. Ratgdo will spin and try to connect to wifi after I manually enter the ssid and credentials (copy-pasting, because I wanted to be _sure_ they were correct). Eventually, it will claim to be connected, but the link to device doesn't work nor do any of the local addresses.

[Log dump](https://gist.github.com/astoltzfus/bf2e0840533e5b81cf530335e9bc9ecf)

abeesix6, 2023-11-06:
Having the same issue. I believe its due to my SSID having a “.” at the end. Is there anyway to get around this?

astoltzfus, 2023-11-06:
I too suspect it is an SSID naming issue. I tried connecting to another network, one without spaces or "."s .. no luck there either (but it _was_ a hotspot, so there's a chance that affected things). I never get an IP address or see any new devices in my router admin panel.

abeesix6, 2023-11-07:
> I too suspect it is an SSID naming issue. I tried connecting to another network, one without spaces or "."s .. no luck there either (but it _was_ a hotspot, so there's a chance that affected things). Either way, I never get an IP address or see any new devices in my router admin panel.

After changing the SSID I was able to connect and set it all up.

ngrusz1, 2023-11-08:
I am having this same issue with my new v2.5 board. Although in my case it usually keeps trying to continuously connect -- it never claims to connect (sometimes after wifi attempt 500, it will say 'max retry reached' but will just keep trying, and sometimes it will find a 169.254.x.x address but that's not a real address). My SSID is relatively simple -- it's a single word with no special characters or spaces, although it does contain both lower and upper case letters.

I'm using the web installer at https://paulwieland.github.io/ratgdo/flash.html to install the firmware; is there a way to confirm it's installing the latest version?

bbbacon12, 2023-11-08:
I am also having this issue with the v2.5 board. I am not convinced it is an SSID problem. My SSID is very simple. It is one word with all lower case letters.

astoltzfus, 2023-11-08:
I've tried several other things now, here's the update:

Normal, MQTT firmware [flashed](https://paulwieland.github.io/ratgdo/flash.html):
- created a new network, with a simple "ratgdo" SSID and non-special character included password .. still unable to connect no matter the version flashed "ratgdo v2.2, Security + 2.0" or "ratgdo v2.26a, Security + 2.0 (secplus 1.0 alpha)"
- flashed ESPHome port
    - this time I was able to _select_ a wifi network from a dropdown instead of having to type it in. I tried connecting to my main 2.4GHz network as well as the simple "ratgdo" (also 2.4GHz) network and neither was able to connect.
    
_Updated original gist with new file containing ESPHome firmware logs_ - [link](https://gist.github.com/astoltzfus/bf2e0840533e5b81cf530335e9bc9ecf#file-ratgdo-esp8266mod-new-ratgdo-network)

I'm beginning to think this might be a hardware issue with this board. Is there any way to confirm or check this? @PaulWieland

ngrusz1, 2023-11-09:
I confirm similar behavior.

Normal, MQTT firmware:
- unable to connect to wifi no matter what wifi settings I use.

ESPHome firmware:
- during installation I am able to select a wifi network from a dropdown. I am able to select my main network, and it does appear to initially connect (the ratgdo is assigned a real IP address from my network, and I am able to access the ratgdo device by pointing my web browser to this address, although this doesn't last). However, the console fills up with "collision detected, waiting to send packet" errors (several per second). I have not yet attempted to connect the device to my Home Assistant server or my garage door, but I assume the console errors are not the expected behavior (some features don't appear to work properly, like resetting the controller).

rlowens, 2023-11-09:
> * However, the console fills up with "collision detected, waiting to send packet" errors (several per second). I have not yet attempted to connect the device to my Home Assistant server or my garage door, but I assume the console errors are not the expected behavior

The expected behavior when not connected to a GDO is "collision detected" errors.

ngrusz1, 2023-11-09:
> The expected behavior when not connected to a GDO is "collision detected" errors.

This would seem to match up with the fact that I am not able to connect to the ratgdo device via the web interface once these errors start appearing. I also noticed some interesting additional behavior when using the normal mqtt firmware. After 500 "unable to connect to wifi" errors the device will issue itself a 169.254.x.x IP address. Interestingly, I was able to connect to the device via that address once, and was presented with a configuration page. I added some values to the configuration, but when I restarted the device I wasn't able to reconnect to it via that IP anymore (despite additional reboots).

Some of these symptoms appear similar to those described in https://github.com/PaulWieland/ratgdo/issues/193 and https://github.com/PaulWieland/ratgdo/issues/195. In at least one case it looks like @PaulWieland advised the user to email directly, perhaps because it was indeed bad hardware? Not sure as there are no additional notes regarding the resolution of the issue.

PaulWieland, 2023-11-09:
@astoltzfus I see that it does eventually connect with the MQTT firmware.
How far away from your access point are you?

LiamThomas7, 2023-11-10:
I was able to resolve this issue. Not sure if this will also resolve it for anyone else having issues. See #214

ngrusz1, 2023-11-10:
> I was able to resolve this issue. Not sure if this will also resolve it for anyone else having issues. See #214

Not sure about the others, but I think this issue is different. When I use the mqtt firmware, I'm actually not able to get the ratgdo to connect to my wifi _at all_. It doesn't seem to "see" my wifi network, it doesn't get a DHCP-issued IP address, and I don't see anything in my router client tables or logs. It eventually assigns itself a 169.254.x.y address (the generic block for automatic private addresses), and there was at least one instance where I was able to point my web browser to that address and visit the configuration page (I'm still trying to figure out how I was even able to do that, because the device never appeared in my router).

Where things do seem to align with your issue is what happened after I was able to reach the config page. I filled out all the requested info, including the username and password for my Mosquitto MQTT server. After rebooting, I was no longer able to access that page, and the ratgdo console filled with ### MQTT DISCONNECTED ### messages (after the prerequisite 500 "unable to connect to wifi" messages. I didn't see anything in my Mosquitto logs, and my router did not list the ratgdo as a connected client.

Just for reference, I've been testing things without actually hooking my ratgdo up to my garage door opener yet. I actually happen to be in the same room as my wireless router, so there definitely shouldn't be a range issue at the moment. The ESPHome firmware seems to be able to find my wireless network (and the networks of a few neighbors), but I run into other problems (as noted above). Not sure if it matters, but my ratgdo is a v2.5 device and appears to use the ESP8266 chip.

PaulWieland, 2023-11-10:
@ngrusz1 please send me an email ratgdo@gmail.com

PaulWieland, 2023-11-10:
@ngrusz1 send me an email at ratgdo@gmail.com

astoltzfus, 2023-11-10:
I was eventually able to get it to connect. It seems like it just takes longer to do so when I'm more than 5-10ft from my router. Now to figure out the proper MQTT username/pw (similar to #214 I think) .. I have different broker, homeassistant/mosquitto mqtt user, and main homeassistant user credentials.

astoltzfus, 2023-11-10:
Re-read documentation, MQTT broker un/pw. Got it all configured.. thank you!

closed 2023-11-10 by astoltzfus