PaulWieland/ratgdo #182: troubleshooting ?

alargeyetti, 2023-11-03:
Any other place I should post for troubleshooting? A forum somewhere?

I can’t figure out what I’m doing wrong. If yall have some advice would very much appreciate it. 

I wired it up as per the diagram. But when I power on the ratgdo my wall button loses power and becomes unresponsive. The garage responds to the MyQ app but does not respond to home assistant commands. It does detect obstructions tho and shows those in home assistant. Additionally, when I send "open" command through home assistant the wall button screen briefly lights up. Same happens with the garage Light command. So something is going on with power delivery to the wall button from the Red wire. The garage does not respond to ratgdo or wall button while it is powered on. When I unplug ratgdo things work normally. 

I included a picture of my wiring below if you notice anything weird. 
I have tried a different usb cable and usb wall wart. And I checked the wiring at the wall button, which is normal R / W?

If I remove the “red” wire from ratgdo the whole system functions normally? And ratgdo sends obstruction notices without issues. 

Here is an image of my wiring. 
[image]

alargeyetti, 2023-11-03:
I figured it out, I am sorry. I am an idiot.
I installed the wrong firmware (2.5 for 2.0 board). Now working perfectly.

PaulWieland, 2023-11-05:
No worries! I will add this to the FAQ.

closed 2023-11-05 by PaulWieland