PaulWieland/ratgdo #183: GDO based power

Dwgystyl, 2023-11-03:
Would it be possible to power the ratgdo via the 12volt signal as the wall button does?
Would possibly save having to run separate power to the d1 ans clean up the overall install.

PaulWieland, 2023-11-05:
No its not possible as there isn't enough overhead provided on the control line. Many logic boards have a 3.3v power source somewhere on the logic board, and thats how I power mine (which is why ratgdo has a 3.3v terminal). Problem is that there are MANY variations of logic boards and not all 3.3v sources work, so I can't really provide step by step instructions to everyone.

closed 2023-11-05 by PaulWieland

rohankapoorcom, 2023-11-06:
@Dwgystyl take a look at this thread where I did some investigation and eventually found a solution https://github.com/PaulWieland/ratgdo/issues/83

Dwgystyl, 2023-11-07:
> @Dwgystyl take a look at this thread where I did some investigation and eventually found a solution #83

That was the idea behind my request,  rather than pulling the MOBO out,  Mainly because they state that the line rests at 12 and is pulled to ground for signalling.. Wonder if using an ldo and supercap would maintain power during these times..   guess there really only is one way to figure that out..

rohankapoorcom, 2023-11-07:
> Wonder if using an ldo and supercap would maintain power during these times.. guess there really only is one way to figure that out..
I saw a thread somewhere where someone tried that. Maybe on the Home Assistant forum? It didn't end up working because it pulled too much current from the opener on the 12v line.

PaulWieland, 2023-11-07:
I'm not saying its impossible, but I've been down this road and gave up. There isn't enough overhead on the 12v line to power esp devices. Additionally, the available overhead depends on the number of wall panels as well as the _type_ of wall panel you have. That's when I threw in the towel and recommended USB power since everyone has an outlet right near the GDO itself. Of course those who are willing can find a 3.3v source on the logic board to power directly from. There are several available on my test rigs, but your mileage may vary. There are MANY different logic board designs.