PaulWieland/ratgdo #186: Sec+ 1.0 on ratgdo 2.0?

senorchris, 2023-11-06:
Hi Paul,

Will Security 1.0 openers be officially supported on ratgdo 2.0 boards?  I ordered a ratgdo last month when issue #62 was looking promising (my order was shipped Oct 12th, I received a 2.0 board).  I flashed it with 2.26 alpha, and after hooking up the board to my 3 terminals, my opener started giving the "3 flash" error code (short in the opener button).

This is my opener, and how I've hooked up the ratgdo:

[image]
[image]

Is there any hope for getting this working with the opener and board I have?

Cheers,

Chris

PaulWieland, 2023-11-06:
Hi Chris,
v2 was officially for security + 2.0 openers only. There is an alpha firmware that adds security +1 support but the hardware can accidentally trigger your door during a reboot, which is not good.

v2.5 hardware addresses this specific problem.

closed 2023-11-06 by PaulWieland

nvllsvm, 2023-11-10:
What changed in v2.5 that fixes the door trigger upon reboot issue? 

I bought a v2.0 shortly before sec+ 1.0 support was added and would like to modify my board.

t-jones14, 2023-11-10:
I have a V2.5 and my door is still opening/closing on reboot. Any ideas?