PaulWieland/ratgdo #188: Wired Ethernet support?

ggiesen, 2023-11-07:
Is there any way to add wired Ethernet support/are there plans to add wired Ethernet support to the ratgdo? 

I just (back)ordered a v2.5 unit for my Chamberlain RJO20C, and I have a general disdain for WiFi home automation devices as I prefer the reliability of a wired connection.

david81, 2023-11-07:
Put me down as interested, as well.

PaulWieland, 2023-11-07:
As of this time it's wifi only, but I agree that a POE solution would be really cool.

ggiesen, 2023-11-07:
Perhaps a board like https://www.olimex.com/Products/IoT/ESP32/ESP32-POE-ISO/open-source-hardware (I'm thinking specifically the SP32-POE-ISO-EA-IND). I realize it's a different physical pinout and spacing than the current WeMos D1 Mini but if the software supported it I could at least hack something up to make the pins work.

rlowens, 2023-11-08:
That Olimex board uses LAN8710A-EZC for ethernet, and [according to this post](https://community.home-assistant.io/t/working-ethernet-thermometer-using-the-olimex-esp-poe-iso-board/175371/14) the [LAN8720 driver in ESPHome](https://esphome.io/components/ethernet.html) works fine with that chip. So, I think that board should work fine with ratgdo-esphome modified to run that ethernet chip. Then just connect wires to the ratgdo pin headers for whatever GPIOs you want to use. Who wants to try?

david81, 2023-11-08:
There are quite a few ethernet boards supported by ESPHome. Do any of them have a matching pinout, by chance? https://esphome.io/components/ethernet.html