PaulWieland/ratgdo #190: How to wire with purple button?

mririgoyen, 2023-11-07:
Hello! Thanks for all your amazing work. I have two purple button openers that have the myQ control pads on the wall. I just got my two v2.5 boards. However, wiring the three wires according to the diagram does not work. I get obstruction status, but I have no control over the door or the light.

There are other pictures that show more than 3 wires going into the board, but I'm having a hard time following what they should actually go to. Can someone who's wired a v2.5 board to a purple button help clarify the correct wiring?

Thank you!

mririgoyen, 2023-11-07:
This is the current wiring I've been trying with no luck.

[image]

PaulWieland, 2023-11-07:
What control protocol did you choose? Can you provide a serial log?

mririgoyen, 2023-11-07:
> What control protocol did you choose? Can you provide a serial log?

I'm not sure I was ever presented with a choice, and I don't see anything in the ESPhome UI. I flashed the chips using: https://github.com/ratgdo/esphome-ratgdo.

mririgoyen, 2023-11-07:
It looks like this might be a limitation of the ESPHome version. I don't currently have a MQTT server stood up to use the other firmware. I will attempt to do that later in the day to confirm if this is truly a wiring issue, or just a software issue.

In the meantime, it would be really cool if the control protocol was configurable in the ESPHome version.

mririgoyen, 2023-11-07:
Confirmed, it is a software issue. My wiring works without a problem using the MQTT firmware. Looking forward to an update to the ESPHome firmware in the future. Thanks again for your great work!

PaulWieland, 2023-11-07:
I just updated the flasher web page for ESP Home to include a note that it is currently only for Security + 2.0.