PaulWieland/ratgdo #191: ratgdo hw and sw v2.5 / secure+ 1.0 / 8500 - unknown value EA

apaperclip, 2023-11-07:
Hi Paul, 

I finally captured the logs of this other unknown value that has been elusive.   When HA logs these I look at the door, light, and obstruction states to see if something is happening to trigger this and I can't find any correlation. 

Based on HA logs this happened at 8:58.   I _think_ this is the corresponding console log.

The last state change before this was the light turning off at 8:50
The next state after this was the light turning back on at 9:05

```
3A58 3900 3A58 3852  | door state: 52 1010010
3A58 3900 3A58 3852  | door state: 52 1010010
3A58 3900 3A58 3852  | door state: 52 1010010
3A58 3900 3A58 3852  | door state: 52 1010010
3A58 3900 3A58 3852  | door state: 52 1010010
3A58 3900 3A58 3852  | door state: 52 1010010
3A58 3900 3A58 3852  | door state: 52 1010010
3A58 3900 3A58 3852  | door state: 52 1010010
3A58 3900 3A58 3852  | door state: 52 1010010
3A58 3900 3A58 3852  | door state: 52 1010010
3A58 3900 3A58 38EA  | door state: EA 11101010
Door state unknown
QUEUE MSG SENT [ratgdo/garage/status/door] 
unknown
3A58 3900 3A58 3852  | door state: 52 1010010
Door state open
QUEUE MSG SENT [ratgdo/garage/status/door] 
open
3A58 3900 3A58 3852  | door state: 52 1010010
3A58 3900 3A58 3852  | door state: 52 1010010
3A58 3900 3A58 3852  | door state: 52 1010010
3A58 3900 3A58 3852  | door state: 52 1010010
3A58 3900 3A58 3852  | door state: 52 1010010
3A58 3900 3A58 3852  | door state: 52 1010010
3A58 3900 3A58 3852  | door state: 52 1010010
3A58 3900 3A58 3852  | door state: 52 1010010
3A58 3900 3A58 3852  | door state: 52 1010010
```

PaulWieland, 2023-11-07:
During this time the door was fully open?

apaperclip, 2023-11-07:
> During this time the door was fully open?

Yes.  -but-  Based on HA logs, unknown can happen the door is closed too, but I can't tell you what the value was from the console. That's been hard to catch.  I need to get remote logs somehow or longer retention of the logs. The webui log viewer seems to die after about 18 hours (could be an old chromebook thing).

apaperclip, 2023-11-09:
I greatly improved my serial logging setup so I have a much longer history and timestamps to coorelate serial log of the value with the state on mqtt.  
I just had a value 5E., and just like EA above, nothing going on before or after. I wonder if this will be cleaned up by the analysis you are doing in the esphome distro with mapping the 3 binary places to state...

PaulWieland, 2023-11-09:
I pushed 2.51 which has new logic for determining door state. 0x5e would map to stopped.

apaperclip, 2023-11-09:
That means there would be a status change going from open to stopped to open, even though there were no actions or state changes physically. That doesn't sound ideal does it?

 I wonder if others on secure+ 1.0  are seeing unknown state errors in HA. Perhaps it something unique to me.  Maybe a bit is flipping for some reason.

PaulWieland, 2023-11-09:
No it certainly doesn't seem right, but based on the testing I've done so far I'm pretty sure that `stopped` is the correct mapping. A bit flip also isn't likely because it's using parity.

None the less I agree that its odd...

apaperclip, 2023-11-09:
Heh, I mean _stopped_ isn't technically wrong, its just been stopped for a while because its closed (or open). Its not a change that's for sure.

ah - I thought we were 8N1, is it using even parity?

Since the 5E there were 4 more,   95, 45, 95, 75.     Looking back over the last couple weeks there is usually between 3 to 7 a day.

PaulWieland, 2023-11-09:
all of those values map to `closed`

PaulWieland, 2023-11-10:
@apaperclip I would like to transfer this issue over to that thread on the ESP Home development in order to consolidate. The same issue will affect both firmware versions. Do you mind posting there and closing down this one? I can't transfer the issue unfortunately.

apaperclip, 2023-11-10:
not at all!  just posted over there.

PaulWieland, 2023-11-15:
Closing as I think we have solved this.

closed 2023-11-15 by PaulWieland